rdiankov/openrave #555: Error happened when ran 'env.StepSimulation(0.001)'

xiaoyuchhusky, 2018-02-14:
When I set the environment's collision checker as 'fcl' and ran 'env.StepSimulation(0.001)', I trapped into the following error loop.

2018-02-14 10:31:50,809 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented
2018-02-14 10:31:50,810 openrave [ERROR] [environment-core.h:2374 _SimulationThread] simulation thread exception: openrave (NotImplemented): not implemented

If I used the default 'ode' checker, the error is gone. Does anyone know why the error happened and how to fix it? I would like to use the 'fcl' checker.

Thank you very much!

Puttichai, 2018-02-14:
Actually the collision checker name for `fcl` should be `fcl_`. Maybe you missed this one.

xiaoyuchhusky, 2018-02-15:
@Puttichai Nice catch! The problem is solved. Thank you very much!

xiaoyuchhusky, 2018-02-15:
@Puttichai I don't know why, it seems the name is fcl in my OpenRave. If I set it as 'fcl_', I will get a [WARN]  that Failed to create name fcl_, interface collisionchecker.