rdiankov/openrave #568: CannotSolveError for Denso robot

ipa-nhg-hd, 2018-03-14:
I am trying to generate IKFast plugin for [Denso robot](https://github.com/ipa-nhg-hd/denso_robot_ros/blob/master/denso_robot_descriptions/vs087_description/urdf/vs087.urdf.xacro). 

After running `$ openrave-robot.py denso.dae --info links`

| name        	| index 	| parents     	|
|-------------	|-------	|-------------	|
| world_frame 	| 0     	|             	|
| base_link   	| 1     	| world_frame 	|
| link_1      	| 2     	| base_link   	|
| link_2      	| 3     	| link_1      	|
| link_3      	| 4     	| link_2      	|
| link_4      	| 5     	| link_3      	|
| link_5      	| 6     	| link_4      	|
| link_6      	| 7     	| link_5      	|
| ee_link     	| 8     	| link_6      	|
| tool0       	| 9     	| link_6      	|
| platform    	| 10    	| world_frame 	|

Finally I run the command:
`python 'openrave-config --python-dir'/openravepy/_openravepy_/ikfast.py --robot=denso.dae --iktype=transform6d --baselink=1 --eelink=9 --savefile=denso_vs087_manipulator_ikfast_solver.cpp`

But this gives me the following error:
```INFO: moved translation [0, 0, 23/200] to right end
INFO: moved translation [0, 0, 79/200] to left end
INFO: moved translation on intersecting axis [0, 0, 43/100] to left
INFO: [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, 79/200]]
INFO: [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
INFO: [[1, 0, 0, 3/100],[0, 0, 1, 0],[0, -1, 0, 0]]
INFO: [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 1, 0, 81/100000000],[-1, 0, 0, -89/200],[0, 0, 1, 0]]
INFO: [[cos(j2), -sin(j2), 0, 0],[sin(j2), cos(j2), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, 1, 43/100],[0, 1, 0, -1/50],[-1, 0, 0, 0]]
INFO: [[cos(j3), -sin(j3), 0, 0],[sin(j3), cos(j3), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, -1, 0],[1, 0, 0, 0],[0, -1, 0, 0]]
INFO: [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 1, 0, 21/100000000],[0, 0, -1, 0],[-1, 0, 0, 0]]
INFO: [[cos(j5), -sin(j5), 0, 0],[sin(j5), cos(j5), 0, 0],[0, 0, 1, 0]]
INFO: [[0, -1, 0, 0],[1, 0, 0, 0],[0, 0, 1, 23/200]]
INFO: ikfast 6d: [j0, j1, j2, j3, j4, j5]
INFO: found 3 consecutive non-intersecting axes links[0:5], vars=[j0, j1, j2]
INFO: found 3 consecutive non-intersecting axes links[2:7], vars=[j1, j2, j3]
INFO: found 3 consecutive non-intersecting axes links[4:9], vars=[j2, j3, j4]
INFO: found 3 consecutive non-intersecting axes links[6:11], vars=[j3, j4, j5]
INFO: try first group 0/4
INFO: computed in 2.403355s
INFO: computed in 1.343992s
INFO: attempting li/woernle/hiller general ik method
INFO: allowed indices: [0]
INFO: assuming equation htj0 is non-zero, dividing by Poly(-cj1*cj2*htj0 + sj1*sj2*htj0 + r22*cj4*htj0 - r21*sj4*cj5*htj0 - r20*sj4*sj5*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r20,r21,r22]')
INFO: not unique: (Poly(cj1*cj2 - sj1*sj2, cj1, sj1, cj2, sj2, htj0, domain='ZZ'), Poly(r22*cj4 - r21*sj4*cj5 - r20*sj4*sj5, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r20,r21,r22]'))
INFO: assuming equation htj0 is non-zero, dividing by Poly(cj1*sj2*htj0**3 - cj1*sj2*htj0 + sj1*cj2*htj0**3 - sj1*cj2*htj0 + r02*cj4*htj0**3 + r02*cj4*htj0 - r01*sj4*cj5*htj0**3 - r01*sj4*cj5*htj0 - r00*sj4*sj5*htj0**3 - r00*sj4*sj5*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r00,r01,r02]')
INFO: not unique: (Poly(-cj1*sj2*htj0**2 + cj1*sj2 - sj1*cj2*htj0**2 + sj1*cj2, cj1, sj1, cj2, sj2, htj0, domain='ZZ'), Poly(r02*cj4*htj0**2 + r02*cj4 - r01*sj4*cj5*htj0**2 - r01*sj4*cj5 - r00*sj4*sj5*htj0**2 - r00*sj4*sj5, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r00,r01,r02]'))
INFO: assuming equation htj0 is non-zero, dividing by Poly(-2*cj1*sj2*htj0**2 - 2*sj1*cj2*htj0**2 + r12*cj4*htj0**3 + r12*cj4*htj0 - r11*sj4*cj5*htj0**3 - r11*sj4*cj5*htj0 - r10*sj4*sj5*htj0**3 - r10*sj4*sj5*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r10,r11,r12]')
INFO: not unique: (Poly(2*cj1*sj2*htj0 + 2*sj1*cj2*htj0, cj1, sj1, cj2, sj2, htj0, domain='ZZ'), Poly(r12*cj4*htj0**2 + r12*cj4 - r11*sj4*cj5*htj0**2 - r11*sj4*cj5 - r10*sj4*sj5*htj0**2 - r10*sj4*sj5, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r10,r11,r12]'))
INFO: assuming equation htj0 is non-zero, dividing by Poly(-43/100*cj1*cj2*htj0 - 1/50*cj1*sj2*htj0 - 89/200*cj1*htj0 - 1/50*sj1*cj2*htj0 + 43/100*sj1*sj2*htj0 + 81/100000000*sj1*htj0 - 21*r21/100000000*cj5*htj0 - 21*r20/100000000*sj5*htj0 + pz*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='QQ[r20,r21,pz]')
INFO: not unique: (Poly(43/100*cj1*cj2 + 1/50*cj1*sj2 + 89/200*cj1 + 1/50*sj1*cj2 - 43/100*sj1*sj2 - 81/100000000*sj1, cj1, sj1, cj2, sj2, htj0, domain='QQ'), Poly(-21*r21/100000000*cj5 - 21*r20/100000000*sj5 + pz, cj4, sj4, cj5, sj5, htj0, domain='QQ[r20,r21,pz]'))
INFO: assuming equation htj0 is non-zero, dividing by Poly(-1/50*cj1*cj2*htj0**3 + 1/50*cj1*cj2*htj0 + 43/100*cj1*sj2*htj0**3 - 43/100*cj1*sj2*htj0 + 81/100000000*cj1*htj0**3 - 81/100000000*cj1*htj0 + 43/100*sj1*cj2*htj0**3 - 43/100*sj1*cj2*htj0 + 1/50*sj1*sj2*htj0**3 - 1/50*sj1*sj2*htj0 + 89/200*sj1*htj0**3 - 89/200*sj1*htj0 - 21*r01/100000000*cj5*htj0**3 - 21*r01/100000000*cj5*htj0 - 21*r00/100000000*sj5*htj0**3 - 21*r00/100000000*sj5*htj0 + (px + 3/100)*htj0**3 + (px - 3/100)*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='QQ[r00,r01,px]')
INFO: not unique: (Poly(1/50*cj1*cj2*htj0**2 - 1/50*cj1*cj2 - 43/100*cj1*sj2*htj0**2 + 43/100*cj1*sj2 - 81/100000000*cj1*htj0**2 + 81/100000000*cj1 - 43/100*sj1*cj2*htj0**2 + 43/100*sj1*cj2 - 1/50*sj1*sj2*htj0**2 + 1/50*sj1*sj2 - 89/200*sj1*htj0**2 + 89/200*sj1 - 3/100*htj0**2, cj1, sj1, cj2, sj2, htj0, domain='QQ'), Poly(-21*r01/100000000*cj5*htj0**2 - 21*r01/100000000*cj5 - 21*r00/100000000*sj5*htj0**2 - 21*r00/100000000*sj5 + px*htj0**2 + px - 3/100, cj4, sj4, cj5, sj5, htj0, domain='QQ[r00,r01,px]'))
INFO: assuming equation htj0 is non-zero, dividing by Poly(1/25*cj1*cj2*htj0**2 - 43/50*cj1*sj2*htj0**2 - 81/50000000*cj1*htj0**2 - 43/50*sj1*cj2*htj0**2 - 1/25*sj1*sj2*htj0**2 - 89/100*sj1*htj0**2 - 21*r11/100000000*cj5*htj0**3 - 21*r11/100000000*cj5*htj0 - 21*r10/100000000*sj5*htj0**3 - 21*r10/100000000*sj5*htj0 + py*htj0**3 - 3/50*htj0**2 + py*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='QQ[r10,r11,py]')
INFO: not unique: [Poly(cj1*cj2 - sj1*sj2, cj1, sj1, cj2, sj2, htj0, domain='ZZ'), Poly(r22*cj4 - r21*sj4*cj5 - r20*sj4*sj5, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r20,r21,r22]')]
INFO: assuming equation htj0 is non-zero, dividing by Poly(-cj1*cj2*htj0 + sj1*sj2*htj0 + r22*cj4*htj0 - r21*sj4*cj5*htj0 - r20*sj4*sj5*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r20,r21,r22]')
INFO: not unique: (Poly(cj1*cj2 - sj1*sj2, cj1, sj1, cj2, sj2, htj0, domain='ZZ'), Poly(r22*cj4 - r21*sj4*cj5 - r20*sj4*sj5, cj4, sj4, cj5, sj5, htj0, domain='ZZ[r20,r21,r22]'))
INFO: not unique: [Poly(43/100*cj1*cj2 + 1/50*cj1*sj2 + 89/200*cj1 + 1/50*sj1*cj2 - 43/100*sj1*sj2 - 81/100000000*sj1, cj1, sj1, cj2, sj2, htj0, domain='QQ'), Poly(-21*r21/100000000*cj5 - 21*r20/100000000*sj5 + pz, cj4, sj4, cj5, sj5, htj0, domain='QQ[r20,r21,pz]')]
INFO: assuming equation htj0 is non-zero, dividing by Poly(-43/100*cj1*cj2*htj0 - 1/50*cj1*sj2*htj0 - 89/200*cj1*htj0 - 1/50*sj1*cj2*htj0 + 43/100*sj1*sj2*htj0 + 81/100000000*sj1*htj0 - 21*r21/100000000*cj5*htj0 - 21*r20/100000000*sj5*htj0 + pz*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='QQ[r20,r21,pz]')
INFO: not unique: (Poly(43/100*cj1*cj2 + 1/50*cj1*sj2 + 89/200*cj1 + 1/50*sj1*cj2 - 43/100*sj1*sj2 - 81/100000000*sj1, cj1, sj1, cj2, sj2, htj0, domain='QQ'), Poly(-21*r21/100000000*cj5 - 21*r20/100000000*sj5 + pz, cj4, sj4, cj5, sj5, htj0, domain='QQ[r20,r21,pz]'))
INFO: assuming equation htj0 is non-zero, dividing by Poly(21*r20/100000000*cj4*cj5*htj0 - 21*r21/100000000*cj4*sj5*htj0 + rxp2_2*cj4*htj0 - rxp1_2*sj4*cj5*htj0 - rxp0_2*sj4*sj5*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='QQ[r20,r21,rxp0_2,rxp1_2,rxp2_2]')
INFO: factors 1 reduced not unique: 21*cj4*cj5*r20/100000000 - 21*cj4*r21*sj5/100000000 + cj4*rxp2_2 - cj5*rxp1_2*sj4 - rxp0_2*sj4*sj5
INFO: assuming equation htj0 is non-zero, dividing by Poly(3/2500*cj1*cj2*htj0 - 129/5000*cj1*sj2*htj0 - 243/5000000000*cj1*htj0 - 129/5000*sj1*cj2*htj0 - 3/2500*sj1*sj2*htj0 - 267/10000*sj1*htj0 - 956749919/2500000000*cj2*htj0 - 89003483/5000000000*sj2*htj0 - 21*npy/50000000*cj5*htj0 - 21*npx/50000000*sj5*htj0 + (pp - 96056250000153/250000000000000)*htj0, cj1, sj1, cj2, sj2, cj4, sj4, cj5, sj5, htj0, domain='QQ[npx,npy,pp]')
[... cut ...]
WARNING: failed with leftvar htj4: CannotSolveError: solveDialytically: more unknowns than equations 12>10
INFO: not all eigenvalues are > 0. min is 0.000000e+00
INFO: not all eigenvalues are > 0. min is 0.000000e+00
INFO: not all eigenvalues are > 0. min is 0.000000e+00
INFO: not all eigenvalues are > 0. min is 0.000000e+00
```

I am running the `latest_stable` branch on Ubuntu 16.04 with ROS kinetic. I have installed sympy-0.7.1 which seems to be a requirement.
I have tried the same method with ABB's IRB2400 and it works fine. Is there something missing in the URDF?

ipa-nhg-hd, 2018-03-19:
I let the script to run for some time, which eventually crashed with a `NoConvergence` error.

Then used this PR: https://github.com/rdiankov/openrave/pull/408, with extraprec=100 and maxsteps=750, I was able to generate the plugin.

closed 2018-03-19 by ipa-nhg-hd

anubhav1772, 2021-07-21:
@ipa-hsd Where to put these(extraprec=100 and maxsteps=750) values?

anubhav1772, 2021-07-24:
I finally found the way to do this. I manually installed the [mpmath](https://github.com/fredrik-johansson/mpmath) with above changes(`extraprec=100 and maxsteps=750`) using `python setup.py install`.  Now, I am able to generate the IK Fast plugin without any problem.

ipa-hsd, 2021-07-26:
Had been a long time since I used this :)

Yan-Xiaodi, 2021-12-17:
> I finally found the way to do this. I manually installed the [mpmath](https://github.com/fredrik-johansson/mpmath) with above changes(`extraprec=100 and maxsteps=750`) using `python setup.py install`. Now, I am able to generate the IK Fast plugin without any problem.

Hello ，I met the same problem with you , so how did you changes the `extraprec=100 and maxsteps=750`;
the `setup.py`  is not  have  this two parameters.