rdiankov/openrave #578: mpmath=0.19.3 can ont find ikfast solution

k-okada, 2018-04-22:
thanks to https://github.com/rdiankov/openrave/pull/531, now we can run `ikfast.py` with ubuntu installed `python-sympy`, but when we run
```
python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot=NextageOpen.rounded.dae --iktype=transform6d --baselink=1 --eelink=15  --savefile=aaa
```
on ubuntu 16.04, we had following error, and `pip install mpmath=0.18` resolves this error. 

```
13*sj14 - r11*sj13*sj14*sj9]
WARNING: not all equations zero, so [j10, j11] vars are not collinear
WARNING: not all equations zero, so [j10, j11] vars are not collinear
WARNING: not all equations zero, so [j10, j11] vars are not collinear
WARNING: not all equations zero, so [j10, j11] vars are not collinear
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 9542, in <module>
    chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2292, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2947, in solveFullIK_6D
    tree = self.solveFullIK_6DGeneral(T0links, T1links, solvejointvars, endbranchtree, usesolvers=1)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3282, in solveFullIK_6DGeneral
    leftovertree = self.SolveAllEquations(AllEquationsExtra,curvars=curvars,othersolvedvars = self.freejointvars+usedvars,solsubs = solsubs,endbranchtree=origendbranchtree)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6726, in SolveAllEquations
    return self.AddSolution(solutions,AllEquations,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 7232, in AddSolution
    nextsolutions[var] = self.SolveAllEquations(AllEquations,curvars=newvars,othersolvedvars=othersolvedvars+[var],solsubs=solsubs+self.Variable(var).subs,endbranchtree=endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6846, in SolveAllEquations
    rawsolutions=self.SolvePairVariables(raweqns,var0,var1,othersolvedvars,unknownvars=curvars+unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 8846, in SolvePairVariables
    return self.SolvePairVariablesHalfAngle(raweqns,var0,var1,othersolvedvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 7759, in SolvePairVariablesHalfAngle
    possiblefinaleq = self.checkFinalEquation(Poly(Malldet,leftvar),subs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 8175, in checkFinalEquation
    roots = mpmath.polyroots(numpy.array(numpy.array(coeffs),numpy.float64))
  File "/usr/lib/python2.7/dist-packages/mpmath/calculus/polynomials.py", line 189, in polyroots
    % maxsteps)
mpmath.libmp.libhyper.NoConvergence: Didn't converge in maxsteps=50 steps.
```

the input dae file can be found at https://gist.github.com/k-okada/83e02785ee2e8a4a14e215cf2376131b

k-okada, 2018-04-22:
humm, system installed sympy (0.7.6) failing too.
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 9587, in <module>
    chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2315, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2970, in solveFullIK_6D
    tree = self.solveFullIK_6DGeneral(T0links, T1links, solvejointvars, endbranchtree, usesolvers=1)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3305, in solveFullIK_6DGeneral
    leftovertree = self.SolveAllEquations(AllEquationsExtra,curvars=curvars,othersolvedvars = self.freejointvars+usedvars,solsubs = solsubs,endbranchtree=origendbranchtree)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6771, in SolveAllEquations
    return self.AddSolution(solutions,AllEquations,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 7575, in AddSolution
    newtree = self.SolveAllEquations(NewEquationsClean,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=newcases, currentcasesubs=newcasesubs, unknownvars=unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6771, in SolveAllEquations
    return self.AddSolution(solutions,AllEquations,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 7277, in AddSolution
    nextsolutions[var] = self.SolveAllEquations(AllEquations,curvars=newvars,othersolvedvars=othersolvedvars+[var],solsubs=solsubs+self.Variable(var).subs,endbranchtree=endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6891, in SolveAllEquations
    rawsolutions=self.SolvePairVariables(raweqns,var0,var1,othersolvedvars,unknownvars=curvars+unknownvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 8891, in SolvePairVariables
    return self.SolvePairVariablesHalfAngle(raweqns,var0,var1,othersolvedvars)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 7804, in SolvePairVariablesHalfAngle
    possiblefinaleq = self.checkFinalEquation(Poly(Malldet,leftvar),subs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 8208, in checkFinalEquation
    if len(coeffs) == 1 and Abs(coeffs[0]) < 2*(10.0**-self.precision):
  File "/usr/lib/python2.7/dist-packages/sympy/core/relational.py", line 103, in __nonzero__
    raise TypeError("cannot determine truth value of\n%s" % self)
TypeError: cannot determine truth value of
Abs(zoo) < 2.0e-8
```

cielavenir, 2024-06-25:
This should be fixed by https://github.com/rdiankov/openrave/pull/1392 , congratz~~~ cc @ntohge @rdiankov

closed 2024-06-25 by cielavenir