rdiankov/openrave #584: Error using ikfast.py

PaulTerrasi, 2018-05-11:
I am getting the following error whenever I try to use ikfast:

```python
No handlers could be found for logger "openravepy.ikfast"
Traceback (most recent call last):
  File "arm_to_cart_target.py", line 34, in <module>
    ikmodel.autogenerate()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/inversekinematics.py", line 653, in autogenerate
    self.generate(iktype=iktype,freejoints=freejoints,precision=precision,forceikbuild=forceikbuild,outputlang=outputlang,ipython=ipython,ikfastmaxcasedepth=ikfastmaxcasedepth)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/inversekinematics.py", line 884, in generate
    chaintree = solver.generateIkSolver(baselink=baselink,eelink=eelink,freeindices=self.freeindices,solvefn=solvefn)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2281, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/inversekinematics.py", line 746, in solveFullIK_6D
    return self.ikfast.IKFastSolver.solveFullIK_6D(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2876, in solveFullIK_6D
    tree = self.TestIntersectingAxes(solvejointvars,Links, LinksInv,endbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2964, in TestIntersectingAxes
    return self.solve6DIntersectingAxes(T0links,T1links,transvars,rotvars,solveRotationFirst=solveRotationFirst, endbranchtree=endbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3137, in solve6DIntersectingAxes
    transtree = self.SolveAllEquations(AllEquations,curvars=curvars,othersolvedvars=othersolvedvars[:],solsubs=solsubs,endbranchtree=newendbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6828, in SolveAllEquations
    rawsolutions=self.SolvePairVariables(raweqns,var0,var1,othersolvedvars,unknownvars=curvars+unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 8955, in SolvePairVariables
    return self.SolvePairVariablesHalfAngle(raweqns,var0,var1,othersolvedvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 7813, in SolvePairVariablesHalfAngle
    Mall, allmonoms = self.solveDialytically(newreducedeqs,ileftvar,returnmatrix=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6146, in solveDialytically
    Mall = [zeros((2*len(dialyticeqs),len(allmonoms))) for i in range(maxdegree+1)]
  File "/home/paul/.local/lib/python2.7/site-packages/sympy/matrices/dense.py", line 1469, in zeros
    return Matrix.zeros(*args, **kwargs)
  File "/home/paul/.local/lib/python2.7/site-packages/sympy/matrices/common.py", line 895, in zeros
    rows, cols = as_int(rows), as_int(cols)
  File "/home/paul/.local/lib/python2.7/site-packages/sympy/core/compatibility.py", line 336, in as_int
    raise ValueError('%s is not an integer' % (n,))
ValueError: (4, 4) is not an integer
```

I'm not sure what could be causing these errors, considering I installed OpenRave following the instructions from https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html, and actually did it on a fresh install of Ubuntu 16.04.

dmcconachie, 2018-05-23:
This could be a problem with sympy; check here:
http://fsuarez6.github.io/blog/workstation-setup-xenial/

PaulTerrasi, 2018-05-23:
My sympy version was the problem. Thank you!

closed 2018-05-23 by PaulTerrasi

zhangzhihao2020, 2021-09-25:
> My sympy version was the problem. Thank you!
I also encountered the same problem. I installed sympy==0.7.1, but still reported an error. What is the version of sympy you installed? Thank you!
```
No handlers could be found for logger "openravepy.ikfast"
Traceback (most recent call last):
  File "test.py", line 15, in <module>
    ikmodel.autogenerate()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/inversekinematics.py", line 658, in autogenerate
    self.generate(iktype=iktype,freejoints=freejoints,precision=precision,forceikbuild=forceikbuild,outputlang=outputlang,ipython=ipython,ikfastmaxcasedepth=ikfastmaxcasedepth)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/inversekinematics.py", line 900, in generate
    chaintree = solver.generateIkSolver(baselink=baselink,eelink=eelink,freeindices=self.freeindices,solvefn=solvefn)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2315, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/databases/inversekinematics.py", line 751, in solveFullIK_6D
    return self.ikfast.IKFastSolver.solveFullIK_6D(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2910, in solveFullIK_6D
    tree = self.TestIntersectingAxes(solvejointvars,Links, LinksInv,endbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2998, in TestIntersectingAxes
    return self.solve6DIntersectingAxes(T0links,T1links,transvars,rotvars,solveRotationFirst=solveRotationFirst, endbranchtree=endbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3162, in solve6DIntersectingAxes
    AllEquations = self.buildEquationsFromPositions(T1links,T1linksinv,transvars,othersolvedvars,uselength=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3677, in buildEquationsFromPositions
    return self.buildEquationsFromTwoSides(Positions,Positionsee,transvars+othersolvedvars,uselength=uselength)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3640, in buildEquationsFromTwoSides
    e = self.SimplifyTransform(self.trigsimp(p2,usedvars)-self.trigsimp(pe2,usedvars))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6431, in SimplifyTransform
    neweq = self._SimplifyRotationDot(eq, self._rotsymbols, self._rotdotgroups)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6552, in _SimplifyRotationDot
    if self.equal(listterms[index0][1], listterms[index1][1]):
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 1570, in equal
    return expand(eq0-eq1) == S.Zero
  File "/home/zzh/.local/lib/python2.7/site-packages/sympy/core/function.py", line 1267, in expand
    return sympify(e).expand(deep=deep, modulus=modulus, **hints)
  File "/home/zzh/.local/lib/python2.7/site-packages/sympy/core/expr.py", line 1884, in expand
    expr = func(deep=deep, **hints)
  File "/home/zzh/.local/lib/python2.7/site-packages/sympy/core/add.py", line 468, in _eval_expand_multinomial
    sargs, terms = self.args, []
KeyboardInterrupt
Segmentation fault (core dumped)
```