rdiankov/openrave #586: IK Solver is missing a lot of valid solutions in my cell layout

jc211, 2018-05-17:
I have a Fanuc robot which has been verified to reach certain points in the real workspace. The IK Solver however can not seem to find the solutions required to get there and marks random areas as unreachable. 
I have attached a python notebook and the robot dae file required to replicate the problem. In the notebook, I manually give the solution to the robot in joint space using SetDOFValues and then I ask the solver to find any solutions at that point. The SetDOFValues completes without complaint and yet the IK solver still can't find the solution. 

Please let me know if I can provide any more details that would help solve the problem. 

I have tried to increase the maxcasedepth on the solver but to no avail. 

[fanuc165R ikfast ticket.tar.gz](https://github.com/rdiankov/openrave/files/2011801/fanuc165R.ikfast.ticket.tar.gz)

sancelot, 2022-02-25:
Have you found a solution or an alternative to this problem ?

jc211, 2022-02-28:
No, sorry, I didn't look into this further

cielavenir, 2022-02-28:
@jc211 so your ipynb is identical to running this in ipython right?

```
import openravepy
import numpy as np
from openravepy import *
env = openravepy.Environment()
collisionChecker = RaveCreateCollisionChecker(env,'ode')
env.SetCollisionChecker(collisionChecker)
env.Reset()
env.Load('fanuc.R2000iC-165R.rounded.dae')
robot = env.GetRobots()[0]
manip = robot.GetActiveManipulator()
T = matrixFromAxisAngle([0,1,0], -120.0*np.pi/180.0)
T[0:3,3] = [-9.696, 1.51, 0.825]
robot.SetTransform(T)
ikmodel = databases.inversekinematics.InverseKinematicsModel(robot,iktype=IkParameterization.Type.Transform6D)
ikmodel.load() or ikmodel.autogenerate()
robot.SetDOFValues([ 0.32114058, -1.37881011, -0.65153141, -4.95656054,  1.35978602, -5.41401134])
sols = manip.FindIKSolutions(manip.GetTransform(), IkFilterOptions.IgnoreEndEffectorCollisions)
```

I was able to get this:

```
In [2]: sols
Out[2]: 
array([[ 0.32105345, -1.37869029, -0.65094246, -1.8148023 , -1.35972887,
        -2.27289856],
       [ 0.32105345, -1.37869029, -0.65094246,  4.46838301, -1.35972887,
        -2.27289856],
       [ 0.32105345, -1.37869029, -0.65094246, -1.8148023 , -1.35972887,
         4.01028675],
       [ 0.32105345, -1.37869029, -0.65094246,  4.46838301, -1.35972887,
         4.01028675],
       [ 0.32114058, -1.37881013, -0.65153143,  1.32662477,  1.35978602,
         0.86917397],
       [ 0.32114058, -1.37881013, -0.65153143, -4.95656054,  1.35978602,
         0.86917397],
       [ 0.32114058, -1.37881013, -0.65153143,  1.32662477,  1.35978602,
        -5.41401134],
       [ 0.32114058, -1.37881013, -0.65153143, -4.95656054,  1.35978602,
        -5.41401134]])
```

jc211, 2022-02-28:
This was posted a long time ago so I don't recall the specifics. Your snippet looks familiar. I remember trying to move the robot on a straight line and only specific points on that line had solutions from the IKSolver. I knew every point on the line segment had a solution and indeed I could manually set the joints to the solutions I found despite IKSolver saying otherwise.