rdiankov/openrave #589: ik solver returns None even where there is a solution

sphanit, 2018-06-05:
Hello,
I have been trying to solve ik Denso vs060 six DOF manipulator. I successfully generated the cpp file but it is not able to provide solutions in most of the cases. I am using Transform6D and for example the following goal transformation is returning None in-spite of being in configuration space:

Tgoal = np.array([[-4.11276647e-01, -9.11510570e-01,  5.03668630e-17,  4.95588540e-01]
                            ,[-9.11510570e-01,  4.11276647e-01,  1.11627850e-16,  1.77862747e-01]
                            ,[-1.22464680e-16,  0.00000000e+00, -1.00000000e+00,  3.93000000e-01]
                             ,[0,0,0,1]])

envrionment setup is found here: https://github.com/crigroup/osr_course_pkgs/tree/master/osr_openrave
I am using pick_and_place.env.xml from the worlds folder.
Can anyone help?

fsuarez6, 2018-07-03:
@sphanit Could you please provide a [MWE](https://en.wikipedia.org/wiki/Minimal_Working_Example).
Moreover, I suggest you to move this issue to the [osr_course_pkgs](https://github.com/crigroup/osr_course_pkgs) repository given that it's most likely related to the pick and place task.

sphanit, 2018-07-03:
@fsuarez6 I would like to close the issue. I was able to solve the problem later. The configuration I provided is most likely out od the workspace or an edge case.

closed 2018-07-03 by sphanit