rdiankov/openrave #590: example error

foolyc, 2018-06-06:
after installed the openrave, running the example, i got the following error
```
~$ openrave.py --example graspplanning
Segmentation fault (core dumped)
~$ openrave.py --example checkvisibility
Segmentation fault (core dumped)

```
i have bullet 2.83.6 and ode 2:0.13.1 in my computer.

it failles the create the env when exe the following code(importing is  all right):
```
from openravepy import *
env = Environment() # create openrave environment
```

foolyc, 2018-06-08:
If i use the master branch,  git reset --hard 9350ebc, another error infos(still memory bugs):
```
~$ openrave.py --example graspplanning
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/graspplanning.py", line 476, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 232, in parseAndCreateThreadedUser
    openravepy_int.RaveInitialize(True)
MemoryError
```

foolyc, 2018-06-18:
```
cmake .. -DOSG_DIR=/usr/local/lib64/ -DOPT_LOG4CXX=OFF
```
```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling OpenRAVE Version 0.9.0, soversion=0.9
-- Using cmake version 3.5.1
-- Using double precision
-- detected system processor: x86_64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.12", minimum required is "2") 
-- installing to /usr/local, python exe /usr/bin/python, python subdir is lib/python2.7/dist-packages, program suffix is '0.9'
-- Performing Test LINKER_HAS_RDYNAMIC
-- Performing Test LINKER_HAS_RDYNAMIC - Success
-- Performing Test LINKER_HAS_BSYMBOLIC
-- Performing Test LINKER_HAS_BSYMBOLIC - Success
-- Performing Test LINKER_HAS_BSYMBOLIC_FUNCTIONS
-- Performing Test LINKER_HAS_BSYMBOLIC_FUNCTIONS - Success
-- Performing Test LINKER_HAS_VISIBILITY
-- Performing Test LINKER_HAS_VISIBILITY - Success
-- Performing Test LINKER_HAS_VISIBILITY_INLINES_HIDDEN
-- Performing Test LINKER_HAS_VISIBILITY_INLINES_HIDDEN - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   regex
--   filesystem
--   system
--   python
--   thread
--   iostreams
--   date_time
--   chrono
--   atomic
-- found boost version: 105800
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmp.so
-- Found GMPXX: /usr/lib/x86_64-linux-gnu/libgmpxx.so
-- Found MPFR: /usr/lib/x86_64-linux-gnu/libmpfr.so
-- Looking for mpfr_div_d in /usr/lib/x86_64-linux-gnu/libmpfr.so
-- Looking for mpfr_div_d in /usr/lib/x86_64-linux-gnu/libmpfr.so - found
-- Could NOT find MPFI
-- Looking for crlibm_init in crlibm
-- Looking for crlibm_init in crlibm - not found
-- Looking for crlibm.h
-- Looking for crlibm.h - not found
-- Using local crlibm
[... 139 lines cut ...]
-- Found OpenThreads: /usr/local/lib64/libOpenThreads.so  
-- Found OpenSceneGraph: /usr/local/lib64/libosgDB.so;/usr/local/lib64/libosgQt.so;/usr/local/lib64/libosgGA.so;/usr/local/lib64/libosgFX.so;/usr/local/lib64/libosgText.so;/usr/local/lib64/libosgViewer.so;/usr/local/lib64/libosgManipulator.so;/usr/local/lib64/libosg.so;/usr/local/lib64/libOpenThreads.so (found suitable version "3.4.2", minimum required is "3.4") 
-- Detected Qt/OSG GUI, making plugin
-- QT dirs: /usr/include/qt4 OPENGL: /usr/include/qt4/QtOpenGL
-- Looking for X11/Xlib.h
-- Looking for X11/Xlib.h - found
-- Link libraries: /usr/lib/x86_64-linux-gnu/libQtDeclarative.so;/usr/lib/x86_64-linux-gnu/libQtGui.so;/usr/lib/x86_64-linux-gnu/libQtCore.so;/usr/lib/x86_64-linux-gnu/libxml2.so;/usr/lib/x86_64-linux-gnu/libQtOpenGL.so;/usr/lib/x86_64-linux-gnu/libGLU.so;/usr/lib/x86_64-linux-gnu/libGL.so;/usr/local/lib64/libosgDB.so;/usr/local/lib64/libosgQt.so;/usr/local/lib64/libosgGA.so;/usr/local/lib64/libosgFX.so;/usr/local/lib64/libosgText.so;/usr/local/lib64/libosgViewer.so;/usr/local/lib64/libosgManipulator.so;/usr/local/lib64/libosg.so;/usr/local/lib64/libOpenThreads.so
-- Detected SoQt/Coin3D GUI, making plugin
-- Looking for dAllocateODEDataForThread
-- Looking for dAllocateODEDataForThread - found
-- Detected ODE multi-threading extensions, but not sure if should enable them... define ODE_USE_MULTITHREAD variable to use them
-- Performing Test NEED_TRIINDEX
-- Performing Test NEED_TRIINDEX - Failed
-- ODE found, building oderave plugin, precision=SINGLE
-- Checking for module 'fcl'
--   Found fcl, version 0.5.0
-- Performing Test FCL_HAS_REPLACEOBJECT
-- Performing Test FCL_HAS_REPLACEOBJECT - Failed
-- Performing Test FCL_SUPPORT_BULK_UPDATE
-- Performing Test FCL_SUPPORT_BULK_UPDATE - Failed
-- Configuring done
-- Generating done
-- Build files have been written to: /home/foolyc/local_install/openrave01/build

```

fsuarez6, 2018-07-03:
@foolyc Did you manage to solve your problem?

If not, try the following:
1. Start OpenRAVE from the command line: `$ openrave`
2. Equivalent to number 1 but using the python bindings:
```python
import openravepy as orpy
env = orpy.Environment()
env.SetDefaultViewer()
```
3. Explicitly use `qtcoin`:
```python
import openravepy as orpy
env = orpy.Environment()
env.SetViewer('qtcoin')
```
> Number 2 and 3 can be executed using IPython (`$ ipython`)

lijiashushu, 2019-10-10:
Hi! Have you soloved this problem? I also met this problem. I don't know why. I installed openrave on ubuntu16.04 following this tutorial https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html .

wuxiangrui188, 2020-04-27:
> Hi! Have you soloved this problem? I also met this problem. I don't know why. I installed openrave on ubuntu16.04 following this tutorial https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html .

老铁，你安装成功了吗？求指教。