rdiankov/openrave #596: TranslationZAxisAngle4D ikfast with SCARA robot  failing

k-okada, 2018-07-10:
Hi
I have 4 Axis SCARA type robot (Yaw, rotate Yaw rotate, Z translate, Yaw rotate) and try to run ikfast with `TranslationZAxisAngle4D` IKType, the program seems working, but if we add '--iktests' option, it shows 
```
openravepy.databases.inversekinematics: testik, success rate: 0.000000, wrong solutions: 0.000000, no solutions: 1.000000, missing solution: 0.000000
```
so I think non of Ik tests working.

If I look the generated ik code more closely, for example `ComputeFk`, It uses `j[3]` instead of `j[2]` at `eetrans[2]=((0.13297)+j[3]);` and `eerot[0]=3.14159265358979;` is constant number.

```
IKFAST_API void ComputeFk(const IkReal* j, IkReal* eetrans, IkReal* eerot) {
for(int dummyiter = 0; dummyiter < 1; ++dummyiter) {
IkReal x0,x1,x2,x3,x4,x5;
x0=IKcos(j[0]);
x1=IKsin(j[0]);
x2=IKcos(j[1]);
x3=IKsin(j[1]);
x4=((0.3)*x0);
x5=((0.3)*x1);
eetrans[0]=(x4+((x2*x4))+(((-1.0)*x3*x5)));
IkReal x6=((1.0)*x5);
eetrans[1]=((((-1.0)*x2*x6))+(((-1.0)*x6))+(((-1.0)*x3*x4)));
eetrans[2]=((0.13297)+j[3]);
eerot[0]=3.14159265358979;
return;
}
IKFAST_ASSERT(0);
}
```

I have put input collada file -> https://gist.githubusercontent.com/k-okada/faf274bfdc9678472affc971208d26d3/raw/255d72f0e1358b52a11ba7bb1826a771759294ef/mxx_scara.round.dae
and output of ikfast.py -> https://gist.github.com/k-okada/faf274bfdc9678472affc971208d26d3#file-gistfile1-txt

If someone know how to fix, please let me know.

P.S. I'm using ikfast.py from production branch.

kanbouchou, 2018-07-10:
Hi,

I think TranslationZAxisAngle4D is not the right type of ik for scara robot for most tasks. For TranslationZAxisAngle4D, angle means angle between base link's z axis and manipulator's z axis which is constant for your robot. 

For scara robots, Translation#AxisAngle$Norm4D should be used most of the time, especially TranslationXAxisAngleZNorm4D for your robot. For Translation#AxisAngle$Norm4D, angle is defined by rotation around $ axis of base link. And for your robot, Z should be the axis around which the angle can be defined.

After doing following in openrave.py
```
manip = robot.GetManipulators()[0]
manip.SetLocalToolDirection([1,0,0])
env.Save('mxx_scara.round.dae')
```

 ik tests are passing
```
openrave.py --database inversekinematics --robot=./mxx_scara.round.dae --usecached --iktests=1000 --iktype=TranslationXAxisAngleZNorm4D
openravepy.databases.inversekinematics: testik, success rate: 1.000000, wrong solutions: 0.000000, no solutions: 0.000000, missing solution: 0.000000
```

k-okada, 2018-07-10:
Thank you for quick response, that's works great! I should read the following error message more carefully.
```
openravepy.ikfast.CannotSolveError: CannotSolveError: cannot solve 4D axis angle IK. Most likely manipulator direction is aligned with the rotation axis

```

closed 2018-07-10 by k-okada