rdiankov/openrave #599: TypeError: argument of type "Poly" is not iterable

LuckyXavier, 2018-08-03:
I'm using the latest stable code and sympy version 0.7.6.1. And I get the error.

INFO: computed in 6.520841s
INFO: computed in 4.544616s
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 9521, in <module>
    chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2281, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2936, in solveFullIK_6D
    tree = self.solveFullIK_6DGeneral(T0links, T1links, solvejointvars, endbranchtree, usesolvers=1)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 3236, in solveFullIK_6DGeneral
    usedinequs = any([var in rawpolyeqs[0][0].gens or var in rawpolyeqs[0][1] for var in self.Variable(solvejointvar).vars])
TypeError: argument of type 'Poly' is not iterable

Thanks

kislaykumarkk, 2019-06-07:
Downgrade sympy to 0.7.1

FelixL-FH, 2022-06-20:
You can download sympy 0.7.3 here [https://github.com/sympy/sympy/releases](sympy). You have to download the version for python 2.
Then extract the .tar.gz file and install with `python setup.py install`
You maybe have to remove mpmath with `sudo apt remove python-mpmath`