rdiankov/openrave #637: unable to install in ubuntu 18

parihardeepanshu, 2019-01-09:


parihardeepanshu, 2019-01-09:
While installing from source , I could'nt install openrave after installing all the dependencies. My command make -j4 after successfully running cmake failed giving the error

Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

sirop, 2019-03-19:
@parihardeepanshu 

You provide too little info.
Please rerun you build with simple `make` as the output of parallel make on several cores often
"conceals" the exact error message.

joao-pm-santos96, 2019-05-13:
For me the error is "error: no matching function for call to ‘min(long unsigned int&, unsigned int&)'". Using ubuntu 18.04 and gcc/g++ version 7.4.0. Any suggestion?

rivertam, 2019-05-23:
The first error I get is:

    /home/ben/openrave/3rdparty/fparser-4.5/extrasrc/fp_identifier_parser.inc:141:15: error: unable to find character literal operator ‘operator""l5’ with ‘char’ argument

followed by a bunch of cryptic errors such as "no matching function call for ... `std::hypot`" which I suspect given my experience with C++ are essentially just cascaded errors from earlier errors.