rdiankov/openrave #638: error while make -j command

lakshmip001, 2019-01-21:
/home/lakshmi/git/openrave/plugins/logging/viewerrecorder.cpp:876:40: error: ‘av_destruct_packet’ was not declared in this scope
                 av_destruct_packet(&pkt);
                                        ^
/home/lakshmi/git/openrave/plugins/logging/viewerrecorder.cpp:880:32: error: ‘av_destruct_packet’ was not declared in this scope
         av_destruct_packet(&pkt);
                                ^
Kindly please let me know what would be the error

cielavenir, 2019-01-21:
I don't have time to take a look at it, but it looks like libavcodec version mismatch. Providing your distribution and libavcodec version would be helpful.

wuxiangrui188, 2020-04-27:
> /home/lakshmi/git/openrave/plugins/logging/viewerrecorder.cpp:876:40: error: ‘av_destruct_packet’ was not declared in this scope
> av_destruct_packet(&pkt);
> ^ [...]

> /home/lakshmi/git/openrave/plugins/logging/viewerrecorder.cpp:876:40: error: ‘av_destruct_packet’ was not declared in this scope
> av_destruct_packet(&pkt);
> ^ [...]

i have the same problem ,have you resolve it ? thanks a lot for give me some idea!
good luck for you