rdiankov/openrave #639: IKFast testik fails with 0% success rate

babakc, 2019-01-23:
I'm building a 5DOF manipulator and haven't had much luck building an IK solver. My urdf for just the manipulator and bash script (to convert to collada, trim to 5 decimals, etc) can be found here: https://github.com/babakc/moveo . The bash script is in the urdf folder. I've stripped out any links/joints that aren't part of the manipulator to simplify things.

I've tried setting a number of different orientations for the joints (so the arm isn't fully stretched out), but continue to get the same error when using the `--iktests` flag:

`openravepy.databases.inversekinematics: testik, success rate: 0.000000, wrong solutions: 0.250000, no solutions: 0.750000, missing solution: 0.000000
openravepy._openravepy_0_9.databases.inversekinematics.InverseKinematicsError: Inverse Kinematics Error
wrong rate 0.250000 > 0!`

I built openrave from commit 9350ebc... Not sure if that's the latest working one (latest commit from master failed to build).

babakc, 2019-01-25:
Hummm, i've gone ahead and created the moveit plugin anyway, and i'm able to move my robot around using the interactive marker, however the EE seems to be stuck in its orientation. Anyone seen this before?

fsuarez6, 2019-01-26:
@babakc: I'm moving our discussion here.
I think the _issue_ is related to the fact that your robot only has 5 DoF, therefore, when you drag the EE in Rviz, it's very unlikely that you are giving a pose that is reachable for your robot.

Something I usually like to test for my IKFast solvers is the following:
1. Generate 1-10K random robot joint configurations (in your case 5 angles). This configuration have to be valid, that means, the robot is not in collision, and the values are within the join limits.
2. For each of the configurations above, compute the FK. That will give you a list of 10K poses for your end-effector.
3. Use the IKFast solver to compute the IK solution for each of the 10K, known, valid, end-effector poses.

That being said, in your case **I would suggest you to use an iterative solver** such as KDL or LMA, both available as kinematics plugins in MoveIt: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/kinematics_configuration/kinematics_configuration_tutorial.html

babakc, 2019-01-26:
@fsuarez6 , thanks. It was my understanding that KDL was really only suitable for => 6dof?

babakc, 2019-01-26:
Okay, this is weird. I changed the last two joints XYZ coords to 0.001, as i noticed i was always getting fails when the message "openravepy.ikfast: solveFullIK_TranslationDirection5D, last 2 axes are intersecting" would appear.

Yet other examples i'd pull off github would be fine. Once i made that slight offset, i'm now getting succesful tests, and i'm now able to move the robot and change orientation of the effector.

openravepy.databases.inversekinematics: testik, success rate: 1.000000, wrong solutions: 0.000000, no solutions: 0.000000, missing solution: 0.230000

fsuarez6, 2019-01-26:
Great!
Regarding KDL, I think it does work with 5 DoF robots.