rdiankov/openrave #650: Couldn't find any package by glob 'openrave0.8-dp-ikfast'

kctoayo88, 2019-02-23:
Hello,

I was trying to install openrave on my PC with Ubuntu 16.04.
I follow this [tutorial](http://docs.ros.org/indigo/api/moveit_ikfast/html/doc/ikfast_tutorial.html) to install
 it.
But when I input  `sudo apt-get install openrave0.8-dp-ikfast`, It show me 

> Couldn't find any package by glob 'openrave0.8-dp-ikfast'

I'm already completed `sudo apt-get update` command before.

So, could you help me to figure out it, maybe because of my internet environment or not ?

cielavenir, 2019-02-23:
http://ppa.launchpad.net/openrave/release/ubuntu/pool/main/o/openrave0.8-dp/ indicates that the package is available for up to precise. I recommend installing from source code...

kctoayo88, 2019-02-25:
thanks, I installed it successfully.

closed 2019-02-25 by kctoayo88