rdiankov/openrave #659: Is Building C examples broken?

sirop, 2019-03-26:
Hallo.

````
cps@debian1:~/openrave/src/cexamples$ git branch
  latest_stable
* master
  production

cps@debian1:~/openrave/src/cexamples$ git log | head -10
commit b1ebe135b4217823ebdf56d9af5fe89b29723603
Merge: 737c4b1b 782a14d7
Author: Rosen Diankov <rosen.diankov@gmail.com>
Date:   Sat Jan 5 09:58:02 2019 +0900

    Merge pull request #632 from PeterBowman/erase-non-const-it

    Consume SampleInfo::itnode as a non-const iterator

commit 782a14d733e52d0507650aa5966c4ef12aec3a4e

cps@debian1:~/openrave/src/cexamples$ cmake .
-- lib openrave0.15_c;openrave0.15-core_c
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cps/openrave/src/cexamples

cps@debian1:~/openrave/src/cexamples$ make
[ 16%] Linking C executable orc_bodyfunctions
/usr/local/lib/libopenrave0.15_c.so: undefined reference to `boost::assertion_failed(char const*, char const*, char const*, long)'collect2: error: ld returned 1 exit status
CMakeFiles/orc_bodyfunctions.dir/build.make:94: recipe for target 'orc_bodyfunctions' failed
make[2]: *** [orc_bodyfunctions] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/orc_bodyfunctions.dir/all' failed
make[1]: *** [CMakeFiles/orc_bodyfunctions.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
````

Building openrave itself was successful.
I am on Debian Stretch, boost version is 1.62 .

Thanks in advance ....

cielavenir, 2019-03-27:
should be fixed by that pull request

sirop, 2019-03-28:
Solved via https://github.com/rdiankov/openrave/pull/662 .

closed 2019-03-28 by sirop