rdiankov/openrave #660: IKFast cannot compute IK in certain orientation

MichalXh, 2019-03-26:
I am playing around with IKFast. I've installed OpenRAVE and generated IKFast executable following [this tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html).

I took robot model for UR5 from [universal_robot repository](https://github.com/ros-industrial/universal_robot)

You can find robot model and scripts I used for debugging [here](https://github.com/MichalXh/debug_ikfast)

So, I've written small python wrapper around binary file to quickly test IKFast. While testing it I've noticed that it cannot find solution for some end effector orientations. You can test it by running `look_for_tooltip_up.py` script from repository linked above. But if I angle the end effector by 1 degree, it works just fine.

I tried to find an explanation for that, and only thing I've found is that if I'll compute bunch of random iks and plot histograms for values of each joint, joints 2 and 4 got very suspicious distribution (you can generate it with `random_poses_histogram.py` script)
[image]

MichalXh, 2019-03-26:
related to #556 #610 #586

DeruiZhao, 2025-02-27:
Hi, I'm having the same problem🥹, do you have a solution for this? @MichalXh