rdiankov/openrave #676: ikfast for yumi

RYANZYC, 2019-04-25:
Hello maintainer!I want to use ikfast instead of KDL algorithm to solve the kinematics of the manipulator.But when I follow the tutorial, the following prompt appears:
[image]
[image]
[image]
Have you configured the ikfast kinematics algorithm for yumi?

tgn3000, 2019-04-25:
Hi RYANZYC! Can you scroll up in your first picture, and upload those several lines that just go before this very first line in the first picture:
> INFO: [[1, 0, 0, -3/100], [0, 0, 1, 17283/100000], [0, -1, 0, 0]]
We would like to see the kinematics matrices for joints j0 and j1. Thank you!

tgn3000, 2019-04-29:
> Hello maintainer!I want to use ikfast instead of KDL algorithm to solve the kinematics of the manipulator.But when I follow the tutorial, the following prompt appears:
> [image]
> [image]
> [image]
> Have you configured the ikfast kinematics algorithm for yumi?

RYANZYC, 2019-04-30:
> 嗨RYANZYC！你可以在你的第一张图片中向上滚动，然后在第一张图片中的第一行之前上传几行：
> 
> > 信息：[[1,0,0，-3 / 100]，[0,0,1,17283 / 100000]，[0，-1,0,0]]
> 
> 我们希望看到关节j0和j1的运动学矩阵。谢谢！

I am sorry that my reply is late.I poured all the models into it, the configuration parameters are as follows：
[image]
[image]
[image]
[image]
[image]

tgn3000, 2019-05-05:
Hi RYANZYC! It seems to me that for your Yumi robot, no consecutive three joint axes (among the seven joints j0, j1, ..., j6) intersect at a point in 3D space, which would cause difficulties for IKFast to generate a solver. That said, I was wondering if you can try setting FREE_INDEX to a different number other than 5 (which refers to j5), that is, one number among 0, 1, 2, 3, 4, or 6. Please let us know. Also, it may be helpful for us to debug if you could share the dae you use. Thanks!

RYANZYC, 2019-05-07:
Hi tgn3000 ! @tgn3000 .I will try to  set different FREE_INDEX,do you have any suggestions for setting up FREE_INDEX?The file is too big for me to upload. I'll send you the xacro format [file.](https://github.com/ethz-asl/yumi/tree/yumi-stacking)

RYANZYC, 2019-05-08:
Hi!tgn3000！ @tgn3000  I find urdf file can be uploaded . This is my urdf.
[yumi.zip](https://github.com/rdiankov/openrave/files/3155040/yumi.zip)

tgn3000, 2019-05-09:
Usually FREE_INDEX=0 would be a good choice, because we can set some value for j0 (discretize [0, 2*pi] and choose different j0 values) and call an IK solver to solve for the rest of six consecutive joints (j1, j2, ..., j6). As for the file, the system on my machine is not compatible with ROS's tools, so I cannot open your urdf file. If convenient, could you please convert the urdf file into the COLLADA .dae file and send me? Thanks!

RYANZYC, 2019-05-09:
Hi!tgn3000！ @tgn3000  I have sent it to your email tgn3000@msn.com. Thank you!

RYANZYC, 2019-05-12:
Hello, when I use world as free_index, index = 0,BASE_LINK="1",EEF_LINK="8".The result are as follow:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
How should I do . Thank you!