rdiankov/openrave #678: undefined symbol: _ZN5boost6system15system_categoryEv

N3k4rk, 2019-05-13:
I have tried to install openrave 0.15, master branch, commit b1ebe13 from Jan 5. The installation went OK (barring issue #641 ). However, after installation, when trying to launch an example:

openrave.py --example graspplanning

I got the error complaining about undefined symbol: _ZN5boost6system15system_categoryEv.

Full text:
$ python program_simple.py
failed to import convexdecompositionpy /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/convexdecompositionpy.so: undefined symbol: _ZN5boost6system15system_categoryEv
Traceback (most recent call last):
  File "program_simple.py", line 2, in <module>
    from openravepy import *
  File "/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py", line 33, in <module>
    openravepy_currentversion = loadstable(__openravepy_version__)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py", line 14, in loadstable
    return _loadversion('_openravepy_'+ver.replace('.', '_'))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/__init__.py", line 19, in _loadversion
    mainpackage = __import__("openravepy", globals(), locals(), [targetname])
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/__init__.py", line 41, in <module>
    from . import databases
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/databases/__init__.py", line 273, in <module>
    import linkstatistics
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/databases/linkstatistics.py", line 72, in <module>
    from .. import pyANN
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/pyANN.py", line 16, in <module>
    from pyANN_int import *
ImportError: /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/pyANN_int.so: undefined symbol: _ZN5boost6system15system_categoryEv

I have tried 3 different configurations. All ended with the same result.
1. Fresh install Ubuntu 18.04 with Boost version 1.65 (googling hinted that the error might be boost-related)
2. Fresh install Ubuntu 16.04 with Boost version 1.58
3. Existing installation Ubuntu 16.04 with Boost 1.58 that has been already running Openrave fine (version 0.9, commit 7c5f5e2).

After installing the new version alongside the old one, I have added version selection to my python code.
So the first two lines were:
__builtins__.__openravepy_version__ = '0.15'
from openravepy import * 

When I set the version to 0.9, openrave launched correctly.

N3k4rk, 2019-10-16:
Possible workaround:
delete these 3 lines from python/bindings/CMakeLists.txt:

if(NOT OPENRAVE_DISABLE_ASSERT_HANDLER)
  add_definitions("-DBOOST_ENABLE_ASSERT_HANDLER") #turns segfault into exception
#endif()

then cmake, make, make install again.

That allowed the example to run in my case. This was with boost 1.58 on ubuntu 18.04. Havent tested with other boost versions. Boost version installed as per instructions from https://github.com/crigroup/openrave-installation (commit c4d32c9) in install-dependencies.sh
But it couldnt download from the address in that script, so I downloaded boost manually from https://sourceforge.net/projects/boost/files/boost/1.58.0/.

cielavenir, 2019-10-17:
@N3k4rk could you check `ldd /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/pyANN_int.so`? If libboost_system is not listed, could you try to change python/bindings/CMakeLists.txt?

before

```
        target_link_libraries(pyANN_int PUBLIC ${STDC_LIBRARY} ANN ${PYTHON_LIBRARIES} ${Boost_PYTHON_LIBRARY} ${Boost_THREAD_LIBRARY} ${extralibs}
                                        PRIVATE boost_assertion_failed)
```

after

```
        target_link_libraries(pyANN_int PUBLIC ${STDC_LIBRARY} ANN ${PYTHON_LIBRARIES} ${Boost_PYTHON_LIBRARY} ${Boost_THREAD_LIBRARY} ${Boost_SYSTEM_LIBRARY} ${extralibs}
                                        PRIVATE boost_assertion_failed)
```

N3k4rk, 2019-10-17:
Default state:
ldd /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/pyANN_int.so
	linux-vdso.so.1 (0x00007ffe6e956000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f979c634000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f979c0b7000)
	libboost_python.so.1.58.0 => /usr/local/lib/libboost_python.so.1.58.0 (0x00007f979be6e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f979bc56000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f979b865000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f979b4c7000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f979ccb1000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f979b2aa000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f979b08b000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f979ae87000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f979ac84000)

After adding boost system to target_link_libraries:
ldd /usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_15/pyANN_int.so
	linux-vdso.so.1 (0x00007ffe351ce000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb9c2e5b000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007fb9c28de000)
	libboost_python.so.1.58.0 => /usr/local/lib/libboost_python.so.1.58.0 (0x00007fb9c2695000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb9c247d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb9c208c000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb9c1cee000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fb9c34d8000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fb9c1ad1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb9c18b2000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fb9c16ae000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fb9c14ab000)

cielavenir, 2019-10-18:
Actually the bad guy is this:

```
root@ubuntu-xenial:/openrave/python/bindings# nm ../../src/libboost_assertion_failed.a|grep category
                 U _ZN5boost6system15system_categoryEv
                 U _ZN5boost6system16generic_categoryEv
```

so, please try adding `#define BOOST_SYSTEM_NO_DEPRECATED` at the top of src/boost_assertion_failed.cpp.

```
// -*- coding: utf-8 -*-
// Copyright (C) 2018 Rosen Diankov
#if !defined(OPENRAVE_DISABLE_ASSERT_HANDLER) && (defined(BOOST_ENABLE_ASSERT_HANDLER))

#define BOOST_SYSTEM_NO_DEPRECATED
#include <openrave/openrave.h>
#include <boost/format.hpp>
```

However why Debian does not require this fix is unknown yet.

\# Well it was lucky that I had ubuntu provisioner in https://github.com/cielavenir/mujin_recruiting .

cielavenir, 2019-10-18:
https://stackoverflow.com/questions/9723793/undefined-reference-to-boostsystemsystem-category-when-compiling

cielavenir, 2019-10-19:
found the reason: https://www.google.com/search?q=as-needed+ubuntu

this roughly means openrave stopped working on ubuntu since boost_assertion_failed was introduced...

rdiankov, 2019-10-20:
@cielavenir did you check that defining `BOOST_SYSTEM_NO_DEPRECATED` will fix the problem?
thanks

cielavenir, 2019-10-20:
@rdiankov I have already checked, but I confirmed just now. It will fix the problem.

closed 2019-10-20 by rdiankov

benthebear93, 2021-06-30:
I got same error while building openrave

Consolidate compiler generated dependencies of target testparabolicramp
[ 80%] Built target qtcoinrave
[ 86%] Built target ikfastsolvers
Consolidate compiler generated dependencies of target rplanners
[ 92%] Built target libopenrave-core
Consolidate compiler generated dependencies of target openrave
Consolidate compiler generated dependencies of target libopenrave-core_c
[ 94%] Linking CXX executable openrave0.62
[ 94%] Built target testparabolicramp
[ 94%] Built target libopenrave-core_c
[ 99%] Built target rplanners
/usr/bin/ld: CMakeFiles/openrave.dir/openrave.cpp.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
//usr/local/lib/libboost_system.so.1.58.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:107: recipe for target 'src/openrave0.62' failed
make[2]: *** [src/openrave0.62] Error 1
CMakeFiles/Makefile2:1010: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

any reason it gives undefined reference to symbol '_ZN5boost6system15system_categoryEv'?