rdiankov/openrave #681: Build errors on aarch64/Ubuntu 16.04

rivertam, 2019-05-28:
The command I'm running is

`cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_BINDINGS=bool:false`

and the error I'm getting is

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp
Run Build Command:"/usr/bin/make" "cmTC_5e46e/fast"
/usr/bin/make -f CMakeFiles/cmTC_5e46e.dir/build.make CMakeFiles/cmTC_5e46e.dir/build
make[1]: Entering directory '/mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_5e46e.dir/CheckSymbolExists.c.o
/usr/bin/cc   -O3 -DNDEBUG   -o CMakeFiles/cmTC_5e46e.dir/CheckSymbolExists.c.o   -c /mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_5e46e
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_5e46e.dir/link.txt --verbose=1
/usr/bin/cc   -O3 -DNDEBUG    -rdynamic CMakeFiles/cmTC_5e46e.dir/CheckSymbolExists.c.o  -o cmTC_5e46e
CMakeFiles/cmTC_5e46e.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text.startup+0x0): undefined reference to `pthread_create'
CheckSymbolExists.c:(.text.startup+0x4): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_5e46e.dir/build.make:97: recipe for target 'cmTC_5e46e' failed
make[1]: *** [cmTC_5e46e] Error 1
make[1]: Leaving directory '/mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_5e46e/fast' failed
make: *** [cmTC_5e46e/fast] Error 2
File /mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>
int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}
Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp
Run Build Command:"/usr/bin/make" "cmTC_39ee2/fast"
/usr/bin/make -f CMakeFiles/cmTC_39ee2.dir/build.make CMakeFiles/cmTC_39ee2.dir/build
make[1]: Entering directory '/mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_39ee2.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create -O3 -DNDEBUG   -o CMakeFiles/cmTC_39ee2.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_39ee2
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_39ee2.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create -O3 -DNDEBUG    -rdynamic CMakeFiles/cmTC_39ee2.dir/CheckFunctionExists.c.o  -o cmTC_39ee2 -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_39ee2.dir/build.make:97: recipe for target 'cmTC_39ee2' failed
make[1]: *** [cmTC_39ee2] Error 1
make[1]: Leaving directory '/mnt/usbdrive/inverse-kinematics/openrave/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_39ee2/fast' failed
make: *** [cmTC_39ee2/fast] Error 2
```

So it looks like cmake isn't properly specifying pthread as a required dependency?

I'm getting this error on both the branch `master` and `latest_stable`. I've looked through the CMakeLists.txt, but I didn't find anywhere specific that seemed relevant.

AndreV84, 2020-04-12:
also an error trying to build it on aarch64: mkdir build && cd build &&cmake .. && make install:
...
`openrave/3rdparty/crlibm-1.0beta4/log-de.c:32:0:
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/double-extended.h:171:2: error: #error "This file should be compiled only for IA32 or IA64 architecture "
 #error "This file should be compiled only for IA32 or IA64 architecture "
  ^~~~~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/double-extended.h:176:2: error: #error "Use icc, version 8.1 or higher to compile for IA64 architecture"
 #error "Use icc, version 8.1 or higher to compile for IA64 architecture"
  ^~~~~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:36:26: error: unknown type name ‘double_ext’; did you mean ‘double’?
 static void log_accurate(double_ext* prh, double_ext* prl, double_ext z, int E, int index) {
                          ^~~~~~~~~~
                          double
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:36:43: error: unknown type name ‘double_ext’; did you mean ‘double’?
 static void log_accurate(double_ext* prh, double_ext* prl, double_ext z, int E, int index) {
                                           ^~~~~~~~~~
                                           double
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:36:60: error: unknown type name ‘double_ext’; did you mean ‘double’?
 static void log_accurate(double_ext* prh, double_ext* prl, double_ext z, int E, int index) {
                                                            ^~~~~~~~~~
                                                            double
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c: In function ‘log_rn’:
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:88:3: error: unknown type name ‘double_ext’; did you mean ‘double’?
   double_ext logirh, r, y, z, th, tl, logde;
   ^~~~~~~~~~
   double
In file included from /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:33:0:
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.h:68:32: error: ‘c’ undeclared (first use in this function); did you mean ‘c7’?
 #define PREFETCH_POLY_QUICK c7=c[0]; c6=c[1]; c5=c[2]; c4=c[3]; c3=c[4]; c2=c[5];
                                ^
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:133:4: note: in expansion of macro ‘PREFETCH_POLY_QUICK’
    PREFETCH_POLY_QUICK; /* defined in log-de.h */
    ^~~~~~~~~~~~~~~~~~~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.h:68:32: note: each undeclared identifier is reported only once for each function it appears in
 #define PREFETCH_POLY_QUICK c7=c[0]; c6=c[1]; c5=c[2]; c4=c[3]; c3=c[4]; c2=c[5];
                                ^
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:133:4: note: in expansion of macro ‘PREFETCH_POLY_QUICK’
    PREFETCH_POLY_QUICK; /* defined in log-de.h */
    ^~~~~~~~~~~~~~~~~~~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:135:9: warning: implicit declaration of function ‘_Asm_getf’; did you mean ‘_IO_getc’? [-Wimplicit-function-declaration]
    i =  _Asm_getf(2/*_FR_D*/, y);  /* Cast y to a 64-bit integer */
         ^~~~~~~~~
         _IO_getc
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:140:55: warning: division by zero [-Wdiv-by-zero]
      if (i<0)                             return (x-x)/0;     /* log(-x) = Nan    */
                                                       ^
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:154:10: warning: implicit declaration of function ‘_Asm_setf’; did you mean ‘scs_set’? [-Wimplicit-function-declaration]
      y = _Asm_setf(2/*_FR_D*/, (i | ULL(3ff0000000000000)) - ULL(0010000000000000) ); /* exponent = -1 */
          ^~~~~~~~~
          scs_set
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:167:11: error: ‘double_ext’ undeclared (first use in this function); did you mean ‘double’?
      r = (double_ext) (argredtable[index].r); /* approx to 1/y.d */
           ^~~~~~~~~~
           double
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:182:14: error: expected ‘;’ before ‘z2’
   double_ext z2,z4, p01, p23, p45, p67, p03, p47,p07;
              ^~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:3: error: ‘z2’ undeclared (first use in this function); did you mean ‘c2’?
   z2  = z*z;      p67 = c6 + z*c7;     p45 = c4 + z*c5;   p23 = c2 + z*c3;    p01 = logirh + z;
   ^~
   c2
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:19: error: ‘p67’ undeclared (first use in this function)
   z2  = z*z;      p67 = c6 + z*c7;     p45 = c4 + z*c5;   p23 = c2 + z*c3;    p01 = logirh + z;
                   ^~~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:40: error: ‘p45’ undeclared (first use in this function)
   z2  = z*z;      p67 = c6 + z*c7;     p45 = c4 + z*c5;   p23 = c2 + z*c3;    p01 = logirh + z;
                                        ^~~
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:59: error: ‘p23’ undeclared (first use in this function)
   z2  = z*z;      p67 = c6 + z*c7;     p45 = c4 + z*c5;   p23 = c2 + z*c3;    p01 = logirh + z;
                                                           ^~~
[... 133 lines cut ...]
   z4  = z2*z2;            p47 = p45 + z2*p67;    p03 = p01 + z2*p23;
   ^~
   z2
/home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:441:27: error: ‘p47’ undeclared (first use in this function); did you mean ‘p45’?
   z4  = z2*z2;            p47 = p45 + z2*p67;    p03 = p01 + z2*p23;
                           ^~~
[... cut ...]
   double_ext c1,c2,c3,c4,c5,c6,c7;
              ^~
3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/build.make:206: recipe for target '3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/log-de.c.o' failed
make[2]: *** [3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/log-de.c.o] Error 1
CMakeFiles/Makefile2:289: recipe for target '3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/all' failed
make[1]: *** [3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
`

43998213, 2021-07-09:
> 尝试在 aarch64 上构建它也是一个错误： mkdir build && cd build &&cmake .. && make install:
> ...
> `openrave/3rdparty/crlibm-1.0beta4/log-de.c:32:0: /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/double-extended.h:171:2: error: #error "This file should be compiled only for IA32 or IA64 architecture " #error "This file should be compiled only for IA32 or IA64 architecture " ^~~~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/double-extended.h:176:2: error: #error "Use icc, version 8.1 or higher to compile for IA64 architecture" #error "Use icc, version 8.1 or higher to compile for IA64 architecture" ^~~~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:36:26: error: unknown type name ‘double_ext’; did you mean ‘double’? static void log_accurate(double_ext* prh, double_ext* prl, double_ext z, int E, int index) { ^~~~~~~~~~ double /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:36:43: error: unknown type name ‘double_ext’; did you mean ‘double’? static void log_accurate(double_ext* prh, double_ext* prl, double_ext z, int E, int index) { ^~~~~~~~~~ double /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:36:60: error: unknown type name ‘double_ext’; did you mean ‘double’? static void log_accurate(double_ext* prh, double_ext* prl, double_ext z, int E, int index) { ^~~~~~~~~~ double /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c: In function ‘log_rn’: /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:88:3: error: unknown type name ‘double_ext’; did you mean ‘double’? double_ext logirh, r, y, z, th, tl, logde; ^~~~~~~~~~ double In file included from /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:33:0: /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.h:68:32: error: ‘c’ undeclared (first use in this function); did you mean ‘c7’? #define PREFETCH_POLY_QUICK c7=c[0]; c6=c[1]; c5=c[2]; c4=c[3]; c3=c[4]; c2=c[5]; ^ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:133:4: note: in expansion of macro ‘PREFETCH_POLY_QUICK’ PREFETCH_POLY_QUICK; /* defined in log-de.h */ ^~~~~~~~~~~~~~~~~~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.h:68:32: note: each undeclared identifier is reported only once for each function it appears in #define PREFETCH_POLY_QUICK c7=c[0]; c6=c[1]; c5=c[2]; c4=c[3]; c3=c[4]; c2=c[5]; ^ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:133:4: note: in expansion of macro ‘PREFETCH_POLY_QUICK’ PREFETCH_POLY_QUICK; /* defined in log-de.h */ ^~~~~~~~~~~~~~~~~~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:135:9: warning: implicit declaration of function ‘_Asm_getf’; did you mean ‘_IO_getc’? [-Wimplicit-function-declaration] i = _Asm_getf(2/*_FR_D*/, y); /* Cast y to a 64-bit integer */ ^~~~~~~~~ _IO_getc /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:140:55: warning: division by zero [-Wdiv-by-zero] if (i<0) return (x-x)/0; /* log(-x) = Nan */ ^ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:154:10: warning: implicit declaration of function ‘_Asm_setf’; did you mean ‘scs_set’? [-Wimplicit-function-declaration] y = _Asm_setf(2/*_FR_D*/, (i | ULL(3ff0000000000000)) - ULL(0010000000000000) ); /* exponent = -1 */ ^~~~~~~~~ scs_set /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:167:11: error: ‘double_ext’ undeclared (first use in this function); did you mean ‘double’? r = (double_ext) (argredtable[index].r); /* approx to 1/y.d */ ^~~~~~~~~~ double /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:182:14: error: expected ‘;’ before ‘z2’ double_ext z2,z4, p01, p23, p45, p67, p03, p47,p07; ^~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:3: error: ‘z2’ undeclared (first use in this function); did you mean ‘c2’? z2 = z*z; p67 = c6 + z*c7; p45 = c4 + z*c5; p23 = c2 + z*c3; p01 = logirh + z; ^~ c2 /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:19: error: ‘p67’ undeclared (first use in this function) z2 = z*z; p67 = c6 + z*c7; p45 = c4 + z*c5; p23 = c2 + z*c3; p01 = logirh + z; ^~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:40: error: ‘p45’ undeclared (first use in this function) z2 = z*z; p67 = c6 + z*c7; p45 = c4 + z*c5; p23 = c2 + z*c3; p01 = logirh + z; ^~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:59: error: ‘p23’ undeclared (first use in this function) z2 = z*z; p67 = c6 + z*c7; p45 = c4 + z*c5; p23 = c2 + z*c3; p01 = logirh + z; ^~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:184:79: error: ‘p01’ undeclared (first use in this function) z2 = z*z; p67 = c6 + z*c7; p45 = c4 + z*c5; p23 = c2 + z*c3; p01 = logirh + z; ^~~ /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:185:3: error: ‘z4’ undeclared (first use in this function); did you mean ‘z2’? z4 = z2*z2; p47 = p45 + z2*p67; p03 = p01 + z2*p23; ^~ z2 /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:185:19: error: ‘p47’ undeclared (first use in this function); did you mean ‘p45’? z4 = z2*z2; p47 = p45 + z2*p67; p03 = p01 + z2*p23; ^~~ p45 /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:185:40: error: ‘p03’ undeclared (first use in this function); did you mean ‘p01’? z4 = z2*z2; p47 = p45 + z2*p67; p03 = p01 + z2*p23; ^~~ p01 /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:186:3: error: ‘p07’ undeclared (first use in this function); did you mean ‘p03’? p07 = p03 + z4*p47; ^~~ p03 In file included from /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/log-de.c:32:0: /home/nvidia/openrave/3rdparty/crlibm-1.0beta4/double-extended.h:189:11: warning: impl [... cut ...] 7; ^~ 3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/build.make:206: recipe for target '3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/log-de.c.o' failed make[2]: *** [3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/log-de.c.o] Error 1 CMakeFiles/Makefile2:289: recipe for target '3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/all' failed make[1]: *** [3rdparty/crlibm-1.0beta4/CMakeFiles/crlibm.dir/all] Error 2 Makefile:129: recipe for target 'all' failed make: *** [all] Error 2 `

any suggestion?

cielavenir, 2021-07-09:
make sure your source tree contains https://github.com/rdiankov/openrave/commit/794d51b367296dac2057f733508eeb0526649ed7

if that does not help, specify -DOPT_ACCURATEMATH=OFF in cmake.

43998213, 2021-07-09:
> make sure your source tree contains [794d51b](https://github.com/rdiankov/openrave/commit/794d51b367296dac2057f733508eeb0526649ed7)
> 
> if that does not help, specify -DOPT_ACCURATEMATH=OFF in cmake.

excellent！ it's working.. but got other error
install on jetson nx arm64

now 
boost 1_67_0 should be 1_58_0?
OpenSceneGraph is 3.6.5
fcl is 0.5.0

i try https://github.com/crigroup/openrave-installation but error always
i want to know explicit version on jetson nx, thank's for generous

cielavenir, 2021-07-09:
if boost is 1.65 or larger is not compatible python - need to use pybind11 version.

you can try https://github.com/cielavenir/openrave-installation/tree/patch-1 but it is tested only on x86_64.

43998213, 2021-07-09:
> if boost is 1.65 or larger is not compatible python - need to use pybind11 version.
> 
> you can try https://github.com/cielavenir/openrave-installation/tree/patch-1 but it is tested only on x86_64.

you must be an angel ,thanks so much，，i can't wait

43998213, 2021-07-09:
> if boost is 1.65 or larger is not compatible python - need to use pybind11 version.
> 
> you can try https://github.com/cielavenir/openrave-installation/tree/patch-1 but it is tested only on x86_64.

openrave-installation-patch-1/boost_1_58_0#  ./b2 -j `nproc`
Performing configuration checks

    - 32-bit                   : no  (cached)
    - 64-bit                   : yes (cached)
    - arm                      : no  (cached)
    - mips1                    : no  (cached)
    - power                    : no  (cached)
    - sparc                    : no  (cached)
    - x86                      : no  (cached)
    - combined                 : no  (cached)

Building the Boost C++ Libraries.

no arm   bu 1.60 ok
openrave-installation-patch-1/boost_1_60_0# ./b2 -j `nproc`
Performing configuration checks

    - 32-bit                   : no
    - 64-bit                   : yes
    - arm                      : yes

Building the Boost C++ Libraries.

should i use 1.60 ?

43998213, 2021-07-09:
i try boost 1.60   openrave-pybind11_archive-stable
root@zjh-desktop:/home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/build# make -j `nproc`
[  1%] Built target ivcon
[  2%] Built target openrave-md5-native
[  3%] Built target openrave-md5
[  9%] Built target ANN
[ 11%] Built target fparser
[ 12%] Built target orcreate_octave
[ 18%] Built target locale
[ 23%] Built target convexdecomposition
[ 23%] Built target orread_octave
[ 26%] Built target PQP
[ 28%] Built target cpp-gen-md5-native
[ 29%] Built target orwrite_octave
[ 30%] Built target interfacehashes_target
[ 31%] Built target boost_assertion_failed
[ 34%] Built target rampoptimizer
[ 45%] Built target libopenrave
[ 46%] Built target baserobots
[ 46%] Built target libopenrave_c
[ 49%] Built target basecontrollers
[ 52%] Built target basesensors
[ 52%] Built target basesamplers
[ 58%] Built target libopenrave-core
[ 60%] Built target configurationcache
[ 62%] Built target grasper
[ 63%] Built target logging
[ 70%] Built target ikfastsolvers
[ 71%] Built target pqprave
[ 73%] Built target rmanipulation
[ 74%] Built target textserver
[ 77%] Built target ParabolicPathSmooth
[ 78%] Built target oderave
[ 79%] Built target fclrave
[ 80%] Linking CXX executable openrave0.28
[ 81%] Built target libopenrave-core_c
[ 89%] Built target qtosgrave
[ 92%] Built target qtcoinrave
[ 92%] Building CXX object plugins/configurationcache/CMakeFiles/openravepy_configurationcache.dir/openravepy_configurationcache.cpp.o
Scanning dependencies of target testparabolicramp
[ 99%] Built target rplanners
[ 99%] Building CXX object plugins/rplanners/ParabolicPathSmooth/CMakeFiles/testparabolicramp.dir/testparabolicramp.cpp.o
libopenrave/libopenrave0.28.so.0.28.0: undefined reference to `boost::filesystem::detail::absolute(boost::filesystem::path const&, boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:111: recipe for target 'src/openrave0.28' failed
make[2]: *** [src/openrave0.28] Error 1
CMakeFiles/Makefile2:1063: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/python2.7/numpy/ndarraytypes.h:1809,
                 from /usr/include/python2.7/numpy/ndarrayobject.h:18,
                 from /usr/include/python2.7/numpy/arrayobject.h:4,
                 from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/openravepy_configurationcache.cpp:17:
/usr/include/python2.7/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
In file included from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/python/bindings/include/openravepy/openravepy_int.h:54,
                 from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/openravepy_configurationcache.cpp:36:
/home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/python/bindings/include/openravepy/bindings.h:55: warning: "FOREACH" redefined
 #define FOREACH(it, v) for(decltype((v).begin()) it = (v).begin(); it != (v).end(); (it)++)
 
In file included from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/configurationcachetree.h:18,
                 from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/openravepy_configurationcache.cpp:15:
/home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/include/openraveplugindefs.h:46: note: this is the location of the previous definition
 #define FOREACH(it, v) for(typeof((v).begin())it = (v).begin(); it != (v).end(); (it)++)
 
In file included from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/python/bindings/include/openravepy/openravepy_int.h:54,
                 from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/openravepy_configurationcache.cpp:36:
/home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/python/bindings/include/openravepy/bindings.h:56: warning: "FOREACH_NOINC" redefined
 #define FOREACH_NOINC(it, v) for(decltype((v).begin()) it = (v).begin(); it != (v).end(); )
 
In file included from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/configurationcachetree.h:18,
                 from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/openravepy_configurationcache.cpp:15:
/home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/include/openraveplugindefs.h:47: note: this is the location of the previous definition
 #define FOREACH_NOINC(it, v) for(typeof((v).begin())it = (v).begin(); it != (v).end(); )
 
In file included from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/python/bindings/include/openravepy/openravepy_int.h:54,
                 from /home/zjh/zjh/ikfast-install/opinstall-a/openrave-pybind11_archive-stable/plugins/configurationcache/openravepy_configurationcache.cpp:36:
[... cut ...]
plugins/rplanners/ParabolicPathSmooth/CMakeFiles/testparabolicramp.dir/build.make:106: recipe for target 'plugins/rplanners/ParabolicPathSmooth/testparabolicramp' failed
make[2]: *** [plugins/rplanners/ParabolicPathSmooth/testparabolicramp] Error 1
CMakeFiles/Makefile2:1769: recipe for target 'plugins/rplanners/ParabolicPathSmooth/CMakeFiles/testparabolicramp.dir/all' failed
make[1]: *** [plugins/rplanners/ParabolicPathSmooth/CMakeFiles/testparabolicramp.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

AndreV84, 2021-07-09:
how can we build on Jetson aarch64?
is it supported?

43998213, 2021-07-09:
> how can we build on Jetson aarch64?
> is it supported?

I've been successful install openrave on Raspberry Pi (ubuntu20)  but jetson be my first time,I also want to know does some body been success?

AndreV84, 2021-07-09:
@43998213 we can kind of upgrade 18.04 to 20.04 on Jetson [ for development purposes, as it is not officially supported], but I am not sure if it will help to install openraven
```
do-release-upgrade
```

AndreV84, 2021-07-09:
tried today with just 
git clone
mkdir build
cd build
cmake -DOPT_ACCURATEMATH=OFF  ..
the output is as below 
```
[ 19%] Building CXX object src/libopenrave/CMakeFiles/libopenrave.dir/trajectory.cpp.o
/home/nvidia/openrave/src/libopenrave/robot.cpp: In member function ‘virtual void OpenRAVE::RobotBase::GripperInfo::SerializeJSON(rapidjson::Value&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, OpenRAVE::dReal, int) const’:
/home/nvidia/openrave/src/libopenrave/robot.cpp:50:56: error: no matching function for call to ‘rapidjson::GenericValue<rapidjson::UTF8<> >::CopyFrom(const Document&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, bool)’
         value.CopyFrom(_docGripperInfo, allocator, true); // need to copy the const strings
                                                        ^
In file included from /home/nvidia/openrave/include/openrave/openrave.h:101:0,
                 from /home/nvidia/openrave/src/libopenrave/libopenrave.h:24,
                 from /home/nvidia/openrave/src/libopenrave/robot.cpp:17:
/usr/include/rapidjson/document.h:805:19: note: candidate: template<class SourceAllocator> rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::CopyFrom(const rapidjson::GenericValue<Encoding, SourceAllocator>&, Allocator&) [with SourceAllocator = SourceAllocator; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]
     GenericValue& CopyFrom(const GenericValue<Encoding, SourceAllocator>& rhs, Allocator& allocator) {
                   ^~~~~~~~
/usr/include/rapidjson/document.h:805:19: note:   template argument deduction/substitution failed:
/home/nvidia/openrave/src/libopenrave/robot.cpp:50:56: note:   candidate expects 2 arguments, 3 provided
         value.CopyFrom(_docGripperInfo, allocator, true); // need to copy the const strings
                                                        ^
/home/nvidia/openrave/src/libopenrave/robot.cpp: In member function ‘virtual void OpenRAVE::RobotBase::GripperInfo::DeserializeJSON(const Value&, OpenRAVE::dReal, int)’:
/home/nvidia/openrave/src/libopenrave/robot.cpp:73:85: error: no matching function for call to ‘rapidjson::GenericDocument<rapidjson::UTF8<> >::CopyFrom(rapidjson::Document&, rapidjson::MemoryPoolAllocator<>&, bool)’
         docGripperInfo.CopyFrom(_docGripperInfo, docGripperInfo.GetAllocator(), true); // need to copy the const strings
                                                                                     ^
In file included from /home/nvidia/openrave/include/openrave/openrave.h:101:0,
                 from /home/nvidia/openrave/src/libopenrave/libopenrave.h:24,
                 from /home/nvidia/openrave/src/libopenrave/robot.cpp:17:
/usr/include/rapidjson/document.h:805:19: note: candidate: template<class SourceAllocator> rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::CopyFrom(const rapidjson::GenericValue<Encoding, SourceAllocator>&, Allocator&) [with SourceAllocator = SourceAllocator; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]
     GenericValue& CopyFrom(const GenericValue<Encoding, SourceAllocator>& rhs, Allocator& allocator) {
                   ^~~~~~~~
/usr/include/rapidjson/document.h:805:19: note:   template argument deduction/substitution failed:
/home/nvidia/openrave/src/libopenrave/robot.cpp:73:85: note:   candidate expects 2 arguments, 3 provided
         docGripperInfo.CopyFrom(_docGripperInfo, docGripperInfo.GetAllocator(), true); // need to copy the const strings
                                                                                     ^
/home/nvidia/openrave/src/libopenrave/robot.cpp: In member function ‘virtual void OpenRAVE::RobotBase::AttachedSensorInfo::SerializeJSON(rapidjson::Value&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, OpenRAVE::dReal, int) const’:
/home/nvidia/openrave/src/libopenrave/robot.cpp:115:68: error: no matching function for call to ‘rapidjson::GenericValue<rapidjson::UTF8<> >::CopyFrom(const Document&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, bool)’
         sensorGeometry.CopyFrom(_docSensorGeometry, allocator, true);
                                                                    ^
In file included from /home/nvidia/openrave/include/openrave/openrave.h:101:0,
                 from /home/nvidia/openrave/src/libopenrave/libopenrave.h:24,
                 from /home/nvidia/openrave/src/libopenrave/robot.cpp:17:
/usr/include/rapidjson/document.h:805:19: note: candidate: template<class SourceAllocator> rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::CopyFrom(const rapidjson::GenericValue<Encoding, SourceAllocator>&, Allocator&) [with SourceAllocator = SourceAllocator; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]
     GenericValue& CopyFrom(const GenericValue<Encoding, SourceAllocator>& rhs, Allocator& allocator) {
                   ^~~~~~~~
/usr/include/rapidjson/document.h:805:19: note:   template argument deduction/substitution failed:
/home/nvidia/openrave/src/libopenrave/robot.cpp:115:68: note:   candidate expects 2 arguments, 3 provided
         sensorGeometry.CopyFrom(_docSensorGeometry, allocator, true);
                                                                    ^
[ 19%] Building CXX object src/libopenrave/CMakeFiles/libopenrave.dir/utils.cpp.o
src/libopenrave/CMakeFiles/libopenrave.dir/build.make:289: recipe for target 'src/libopenrave/CMakeFiles/libopenrave.dir/robot.cpp.o' failed
make[2]: *** [src/libopenrave/CMakeFiles/libopenrave.dir/robot.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1334: recipe for target 'src/libopenrave/CMakeFiles/libopenrave.dir/all' failed
make[1]: *** [src/libopenrave/CMakeFiles/libopenrave.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

43998213, 2021-07-09:
> @43998213 we can kind of upgrade 18.04 to 20.04 on Jetson [ for development purposes, as it is not officially supported], but I am not sure if it will help to install openrave
> 
> ```
> do-release-upgrade
> ```

do-release-upgrade on my jetson failed，bu if you can fix it ,you should try。

https://github.com/cielavenir/openrave-installation/tree/patch-1
it should be worked, God bless us

AndreV84, 2021-07-09:
what was the error? for the upgrade failure?

AndreV84, 2021-07-09:
you would have to purge chromium-browser-l10n, chromium-browser, also libopencv-dev* then it will get though highly likely

43998213, 2021-07-09:
> you would have to purge chromium-browser-l10n, chromium-browser, also libopencv-dev* then it will get through highly likely

Thanks, It's a great idea, I should try it

AndreV84, 2021-07-09:
I tried the script ./install-osg
https://github.com/cielavenir/openrave-installation/blob/master/install-osg.sh
```
[ 14%] Building CXX object src/osgDB/CMakeFiles/osgDB.dir/ConvertBase64.cpp.o
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp: In function ‘int osgDB::base64_decode_value(char)’:
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
         static const char decoding[] = {62,-1,-1,-1,63,52,53,54,55,56,57,58,59,60,61,-1,-1,-1,-2,-1,-1,-1,0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,-1,-1,-1,-1,-1,-1,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51};
                                                                                                                                                                                                                                                                              ^
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-2’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
src/osgDB/CMakeFiles/osgDB.dir/build.make:185: recipe for target 'src/osgDB/CMakeFiles/osgDB.dir/ConvertBase64.cpp.o' failed
make[2]: *** [src/osgDB/CMakeFiles/osgDB.dir/ConvertBase64.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

AndreV84, 2021-07-09:
yes, I've run the same scripts on 20.04 jetson
lets see if it works with these scripts: https://github.com/cielavenir/openrave-installation/archive/refs/heads/patch-1.zip

43998213, 2021-07-09:
> 是的，我在 20.04 jetson 上运行了相同的脚本，
> 让我们看看它是否适用于这些脚本：[https](https://github.com/cielavenir/openrave-installation/archive/refs/heads/patch-1.zip) : [//github.com/cielavenir/openrave-installation/archive/refs/heads/patch-1.zip](https://github.com/cielavenir/openrave-installation/archive/refs/heads/patch-1.zip)

good job, wish you success

AndreV84, 2021-07-09:
> good job, wish you success

unfortunately failed at this point
```
[ 13%] Built target osgUtil
[ 13%] Building CXX object src/osgDB/CMakeFiles/osgDB.dir/ConvertBase64.cpp.o
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp: In function ‘int osgDB::base64_decode_value(char)’:
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
   31 | 1,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51};
      |                                                              ^

/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-2’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
/home/nvidia/git/OpenSceneGraph/src/osgDB/ConvertBase64.cpp:31:270: error: narrowing conversion of ‘-1’ from ‘int’ to ‘char’ [-Wnarrowing]
make[2]: *** [src/osgDB/CMakeFiles/osgDB.dir/build.make:167: src/osgDB/CMakeFiles/osgDB.dir/ConvertBase64.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2750: src/osgDB/CMakeFiles/osgDB.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
Scanning dependencies of target install_ld_conf
Copying openscenegraph.conf to /etc/ld.so.conf.d and running ldconfig
Built target install_ld_conf
```

AndreV84, 2021-07-09:
However it seems possible to install osg manually with 
```
git clone https://github.com/openscenegraph/OpenSceneGraph
cd OpenSceneGraph
cmake .
make -j6
sudo make install
```

AndreV84, 2021-07-09:
from script getting this
```
-- Found Threads: TRUE  
-- Could NOT find Boost (missing: regex filesystem system python thread iostreams date_time) (found version "1.58.0")
-- found boost version: 105800
CMake Error at CMakeLists.txt:357 (message):
  boost thread and date_time libraries are necessary

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/git/openrave/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/git/openrave/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.

```
without the script but from github sources getting this:
                                                             ^
/home/nvidia/openrave/build/openrave/3rdparty/ann/src/perf.cpp:128:61: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   print_one_stat("    average_error    ", ann_average_err, 1);
                                                             ^
/home/nvidia/openrave/build/openrave/3rdparty/ann/src/perf.cpp:129:58: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
   print_one_stat("    rank_error       ", ann_rank_err, 1);
                                                          ^
[ 23%] Linking CXX static library libANN.a
[ 23%] Built target ANN
[ 24%] Linking CXX executable cpp-gen-md5-native
[ 24%] Built target cpp-gen-md5-native
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```
[on 18,04 system] with osg installed from github, as script failed
```

AndreV84, 2021-07-09:
after using this  https://github.com/Tencent/rapidjson instead of apt package version it got further with 
```

```

so the outcome is on 18.04 system as follows
```
87%] Linking CXX shared library libopenrave0.62-core_c.so
[ 87%] Built target libopenrave-core_c
Scanning dependencies of target rplanners
In file included from /usr/include/python2.7/numpy/ndarraytypes.h:1809:0,
                 from /usr/include/python2.7/numpy/ndarrayobject.h:18,
                 from /usr/include/python2.7/numpy/arrayobject.h:4,
                 from /home/nvidia/openrave/python/bindings/include/openravepy/openravepy_int.h:52,
                 from /home/nvidia/openrave/python/bindings/include/openravepy/openravepy_jointinfo.h:21,
                 from /home/nvidia/openrave/python/bindings/openravepy_kinbody.cpp:18:
/usr/include/python2.7/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
[ 87%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/constraintparabolicsmoother.cpp.o
[ 87%] Linking CXX executable openrave0.62
libopenrave-core/libopenrave0.62-core.so.0.62.2: undefined reference to `boost::this_thread::hiden::sleep_until(timespec const&)'
libopenrave/libopenrave0.62.so.0.62.2: undefined reference to `boost::filesystem::path::m_normalize()'
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:116: recipe for target 'src/openrave0.62' failed
make[2]: *** [src/openrave0.62] Error 1
CMakeFiles/Makefile2:1154: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 87%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/cubicretimer.cpp.o
In file included from /usr/include/python2.7/numpy/ndarraytypes.h:1809:0,
                 from /usr/include/python2.7/numpy/ndarrayobject.h:18,
                 from /usr/include/python2.7/numpy/arrayobject.h:4,
                 from /home/nvidia/openrave/python/bindings/include/openravepy/openravepy_int.h:52,
                 from /home/nvidia/openrave/python/bindings/include/openravepy/openravepy_kinbody.h:20,
                 from /home/nvidia/openrave/python/bindings/openravepy_robot.cpp:18:
/usr/include/python2.7/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^~~~~~~
[ 87%] Building CXX object python/bindings/CMakeFiles/openravepy_int.dir/openravepy_global.cpp.o
```
then it fgails at the end though
```
[ 99%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/parabolicsmoother2.cpp.o
In file included from /usr/include/python2.7/numpy/ndarrayobject.h:27:0,
                 from /usr/include/python2.7/numpy/arrayobject.h:4,
                 from /home/nvidia/openrave/python/bindings/include/openravepy/bindings.h:25,
                 from /home/nvidia/openrave/python/bindings/bindings.cpp:16:
/usr/include/python2.7/numpy/__multiarray_api.h:1453:1: warning: ‘int _import_array()’ defined but not used [-Wunused-function]
 _import_array(void)
 ^~~~~~~~~~~~~
/home/nvidia/openrave/plugins/rplanners/parabolicsmoother2.cpp: In member function ‘virtual OpenRAVE::RampOptimizerInternal::CheckReturn rplanners::ParabolicSmoother2::MyRampNDFeasibilityChecker::Check2(const std::vector<OpenRAVE::RampOptimizerInternal::RampND>&, int, std::vector<OpenRAVE::RampOptimizerInternal::RampND>&)’:
/home/nvidia/openrave/plugins/rplanners/parabolicsmoother2.cpp:66:33: warning: unused variable ‘vswitchtimes’ [-Wunused-variable]
             std::vector<dReal> &vswitchtimes=_vswitchtimes;
                                 ^~~~~~~~~~~~
[ 99%] Linking CXX shared library openravepy_int.so
[ 99%] Linking CXX shared library librplanners.so
[ 99%] Built target rplanners
[ 99%] Built target openravepy_int
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

AndreV84, 2021-07-09:
with another libboost from apt repository it ends up like this
```
             ^~~~~~~~~~~~~~~
[ 91%] Building CXX object 3rdparty/collada-2.4.0/CMakeFiles/collada15reader.dir/src/modules/stdErrPlugin/stdErrPlugin.cpp.o
In file included from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeURI.cpp:10:0:
/home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae.h:290:10: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
     std::auto_ptr<charEncoding> localCharEncoding;
          ^~~~~~~~
In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
                 from /usr/include/c++/7/locale:43,
                 from /usr/local/include/boost/filesystem/path_traits.hpp:29,
                 from /usr/local/include/boost/filesystem/path.hpp:25,
                 from /usr/local/include/boost/filesystem/operations.hpp:25,
                 from /usr/local/include/boost/filesystem/convenience.hpp:22,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae.h:28,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeURI.cpp:10:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeURI.cpp:14:0:
/home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae/daeErrorHandler.h:58:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
  static std::auto_ptr<daeErrorHandler> _defaultInstance;
              ^~~~~~~~
In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
                 from /usr/include/c++/7/locale:43,
                 from /usr/local/include/boost/filesystem/path_traits.hpp:29,
                 from /usr/local/include/boost/filesystem/path.hpp:25,
                 from /usr/local/include/boost/filesystem/operations.hpp:25,
                 from /usr/local/include/boost/filesystem/convenience.hpp:22,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae.h:28,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeURI.cpp:10:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeZAEUncompressHandler.cpp:3:0:
/home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae.h:290:10: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
     std::auto_ptr<charEncoding> localCharEncoding;
          ^~~~~~~~
In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
                 from /usr/include/c++/7/locale:43,
                 from /usr/local/include/boost/filesystem/path_traits.hpp:29,
                 from /usr/local/include/boost/filesystem/path.hpp:25,
                 from /usr/local/include/boost/filesystem/operations.hpp:25,
                 from /usr/local/include/boost/filesystem/convenience.hpp:22,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae.h:28,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeZAEUncompressHandler.cpp:3:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeZAEUncompressHandler.cpp:4:0:
/home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae/daeErrorHandler.h:58:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
  static std::auto_ptr<daeErrorHandler> _defaultInstance;
              ^~~~~~~~
In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
                 from /usr/include/c++/7/locale:43,
                 from /usr/local/include/boost/filesystem/path_traits.hpp:29,
                 from /usr/local/include/boost/filesystem/path.hpp:25,
                 from /usr/local/include/boost/filesystem/operations.hpp:25,
                 from /usr/local/include/boost/filesystem/convenience.hpp:22,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae.h:28,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/dae/daeZAEUncompressHandler.cpp:3:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/modules/stdErrPlugin.h:13:0,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/modules/stdErrPlugin/stdErrPlugin.cpp:9:
/home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae/daeErrorHandler.h:58:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
  static std::auto_ptr<daeErrorHandler> _defaultInstance;
              ^~~~~~~~
In file included from /usr/include/c++/7/memory:80:0,
[... 36 lines cut ...]
In file included from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/dae/daeDatabase.h:14:0,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/include/modules/daeSTLDatabase.h:21,
                 from /home/nvidia/openrave/3rdparty/collada-2.4.0/src/modules/STLDatabase/daeSTLDatabase.cpp:9:
[... cut ...]
   template<typename> class auto_ptr;
                            ^~~~~~~~
[ 91%] Linking CXX static library libcollada15reader.a
[ 91%] Built target collada15reader
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```
given the command executed is 
```
 cmake -DOPT_ACCURATEMATH=OFF  -DBoost_NO_SYSTEM_PATHS=FALSE -DBOOST_ROOT=/usr/lib/aarch64-linux-gnu -DODE_USE_MULTITHREAD=ON -DCMAKE_CXX_STANDARD=11  ..
```
with  installed this
```
sudo apt-get install libboost-dev libboost-all-dev

AndreV84, 2021-07-09:
another attempt
```
                            ^~~~~~~~
[ 67%] Building CXX object src/libopenrave-core/CMakeFiles/libopenrave-core.dir/xfileparser/XFileParser.cpp.o
[ 67%] Building CXX object src/libopenrave-core/CMakeFiles/libopenrave-core.dir/xfileparser/XFileBindings.cpp.o
[ 67%] Linking CXX shared library libopenrave0.62-core.so
../../3rdparty/collada-2.4.0/libcollada15reader.a(daeUtils.cpp.o): In function `cdom::getRandomFileName[abi:cxx11]()':
daeUtils.cpp:(.text+0xcc0): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
[ 67%] Built target libopenrave-core
Scanning dependencies of target openrave
[ 67%] Building CXX object src/CMakeFiles/openrave.dir/openrave.cpp.o
[ 68%] Linking CXX executable openrave0.62
libopenrave-core/libopenrave0.62-core.so.0.62.2: undefined reference to `boost::this_thread::hiden::sleep_until(timespec const&)'
libopenrave/libopenrave0.62.so.0.62.2: undefined reference to `boost::filesystem::path::m_normalize()'
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:116: recipe for target 'src/openrave0.62' failed
make[2]: *** [src/openrave0.62] Error 1
CMakeFiles/Makefile2:1202: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

AndreV84, 2021-07-09:
it seems on 20.04 the script installation got through 
osg did not getr through though
but seems all built  finally 100%

43998213, 2021-07-12:
> it seems on 20.04 the script installation got through
> osg did not getr through though
> but seems all built finally 100%

I tried upgrade to 20.04 ,bu snap failed ,I'm going to try jetson nano

AndreV84, 2021-07-12:
you would have to use do-release-upgrade
any jetson including nano will do, but the latter will be the slowest

43998213, 2021-07-13:
> you would have to use do-release-upgrade
> any jetson including nano will do, but the latter will be the slowest

Our country's network may be constrained ，they told me network problem。

I tried install Boost 1_58 on jetson nano with ./bjam install toolset=gcc-arm --prefix=/usr/local/boost
but:
    "/usr/bin/arm-linux-gnueabihf-gcc"  -ftemplate-depth-128 -O3 -finline-functions -Wno-inline -Wall -pthread -fPIC  -DBOOST_ALL_NO_LIB=1 -DBOOST_PYTHON_SOURCE -DNDEBUG  -I"." -I"/usr/include/python2.7" -c -o "bin.v2/libs/python/build/gcc-arm/release/threading-multi/exec.o" "libs/python/src/exec.cpp"

...failed gcc.compile.c++ bin.v2/libs/python/build/gcc-arm/release/threading-multi/exec.o...
gcc.compile.c++ bin.v2/libs/python/build/gcc-arm/release/threading-multi/object/function_doc_signature.o
In file included from ./boost/python/detail/wrap_python.hpp:50:0,
                 from ./boost/python/detail/prefix.hpp:13,
                 from ./boost/python/converter/registrations.hpp:8,
                 from libs/python/src/object/function_doc_signature.cpp:9:
/usr/include/python2.7/pyconfig.h:24:12: fatal error: arm-linux-gnueabihf/python2.7/pyconfig.h:
 #  include <arm-linux-gnueabihf/python2.7/pyconfig.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

    "/usr/bin/arm-linux-gnueabihf-gcc"  -ftemplate-depth-128 -O3 -finline-functions -Wno-inline -Wall -pthread -fPIC  -DBOOST_ALL_NO_LIB=1 -DBOOST_PYTHON_SOURCE -DNDEBUG  -I"." -I"/usr/include/python2.7" -c -o "bin.v2/libs/python/build/gcc-arm/release/threading-multi/object/function_doc_signature.o" "libs/python/src/object/function_doc_signature.cpp"

...failed gcc.compile.c++ bin.v2/libs/python/build/gcc-arm/release/threading-multi/object/function_doc_signature.o...
...skipped <pbin.v2/libs/python/build/gcc-arm/release/threading-multi>libboost_python.so.1.58.0 for lack of <pbin.v2/libs/python/build/gcc-arm/release/threading-multi>numeric.o...
...skipped <p/usr/local/boost/lib>libboost_python.so.1.58.0 for lack of <pbin.v2/libs/python/build/gcc-arm/release/threading-multi>libboost_python.so.1.58.0...
...skipped <p/usr/local/boost/lib>libboost_python.so for lack of <p/usr/local/boost/lib>libboost_python.so.1.58.0...
...failed updating 57 targets...
...skipped 8 targets...

AndreV84, 2021-07-15:
yeah, on 18.04 Ubuntu it won't work, as per my experience

spullil150, 2022-05-13:
> tried today with just
> 
> ```
> git clone https://github.com/rdiankov/openrave/
> cd openrave
> mkdir build
> cd build
> apt-get install rapidjson-dev -y
> cmake -DOPT_ACCURATEMATH=OFF  ..
> ```
> 
> the output is as below
> 
> ```
> [ 19%] Building CXX object src/libopenrave/CMakeFiles/libopenrave.dir/trajectory.cpp.o
> /home/nvidia/openrave/src/libopenrave/robot.cpp: In member function ‘virtual void OpenRAVE::RobotBase::GripperInfo::SerializeJSON(rapidjson::Value&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, OpenRAVE::dReal, int) const’:
> /home/nvidia/openrave/src/libopenrave/robot.cpp:50:56: error: no matching function for call to ‘rapidjson::GenericValue<rapidjson::UTF8<> >::CopyFrom(const Document&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, bool)’
>          value.CopyFrom(_docGripperInfo, allocator, true); // need to copy the const strings
>                                                         ^
> In file included from /home/nvidia/openrave/include/openrave/openrave.h:101:0,
>                  from /home/nvidia/openrave/src/libopenrave/libopenrave.h:24,
>                  from /home/nvidia/openrave/src/libopenrave/robot.cpp:17:
> /usr/include/rapidjson/document.h:805:19: note: candidate: template<class SourceAllocator> rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::CopyFrom(const rapidjson::GenericValue<Encoding, SourceAllocator>&, Allocator&) [with SourceAllocator = SourceAllocator; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]
>      GenericValue& CopyFrom(const GenericValue<Encoding, SourceAllocator>& rhs, Allocator& allocator) {
>                    ^~~~~~~~
> /usr/include/rapidjson/document.h:805:19: note:   template argument deduction/substitution failed:
> /home/nvidia/openrave/src/libopenrave/robot.cpp:50:56: note:   candidate expects 2 arguments, 3 provided
>          value.CopyFrom(_docGripperInfo, allocator, true); // need to copy the const strings
>                                                         ^
> /home/nvidia/openrave/src/libopenrave/robot.cpp: In member function ‘virtual void OpenRAVE::RobotBase::GripperInfo::DeserializeJSON(const Value&, OpenRAVE::dReal, int)’:
> /home/nvidia/openrave/src/libopenrave/robot.cpp:73:85: error: no matching function for call to ‘rapidjson::GenericDocument<rapidjson::UTF8<> >::CopyFrom(rapidjson::Document&, rapidjson::MemoryPoolAllocator<>&, bool)’
>          docGripperInfo.CopyFrom(_docGripperInfo, docGripperInfo.GetAllocator(), true); // need to copy the const strings
>                                                                                      ^
> In file included from /home/nvidia/openrave/include/openrave/openrave.h:101:0,
>                  from /home/nvidia/openrave/src/libopenrave/libopenrave.h:24,
>                  from /home/nvidia/openrave/src/libopenrave/robot.cpp:17:
> /usr/include/rapidjson/document.h:805:19: note: candidate: template<class SourceAllocator> rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::CopyFrom(const rapidjson::GenericValue<Encoding, SourceAllocator>&, Allocator&) [with SourceAllocator = SourceAllocator; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]
>      GenericValue& CopyFrom(const GenericValue<Encoding, SourceAllocator>& rhs, Allocator& allocator) {
>                    ^~~~~~~~
> /usr/include/rapidjson/document.h:805:19: note:   template argument deduction/substitution failed:
> /home/nvidia/openrave/src/libopenrave/robot.cpp:73:85: note:   candidate expects 2 arguments, 3 provided
>          docGripperInfo.CopyFrom(_docGripperInfo, docGripperInfo.GetAllocator(), true); // need to copy the const strings
>                                                                                      ^
> /home/nvidia/openrave/src/libopenrave/robot.cpp: In member function ‘virtual void OpenRAVE::RobotBase::AttachedSensorInfo::SerializeJSON(rapidjson::Value&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, OpenRAVE::dReal, int) const’:
> /home/nvidia/openrave/src/libopenrave/robot.cpp:115:68: error: no matching function for call to ‘rapidjson::GenericValue<rapidjson::UTF8<> >::CopyFrom(const Document&, rapidjson::GenericDocument<rapidjson::UTF8<> >::AllocatorType&, bool)’
>          sensorGeometry.CopyFrom(_docSensorGeometry, allocator, true);
>                                                                     ^
> In file included from /home/nvidia/openrave/include/openrave/openrave.h:101:0,
>                  from /home/nvidia/openrave/src/libopenrave/libopenrave.h:24,
>                  from /home/nvidia/openrave/src/libopenrave/robot.cpp:17:
> /usr/include/rapidjson/document.h:805:19: note: candidate: template<class SourceAllocator> rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::CopyFrom(const rapidjson::GenericValue<Encoding, SourceAllocator>&, Allocator&) [with SourceAllocator = SourceAllocator; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]
>      GenericValue& CopyFrom(const GenericValue<Encoding, SourceAllocator>& rhs, Allocator& allocator) {
>                    ^~~~~~~~
> /usr/include/rapidjson/document.h:805:19: note:   template argument deduction/substitution failed:
> /home/nvidia/openrave/src/libopenrave/robot.cpp:115:68: note:   candidate expects 2 arguments, 3 provided
[... cut ...]
> src/libopenrave/CMakeFiles/libopenrave.dir/build.make:289: recipe for target 'src/libopenrave/CMakeFiles/libopenrave.dir/robot.cpp.o' failed
> make[2]: *** [src/libopenrave/CMakeFiles/libopenrave.dir/robot.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> CMakeFiles/Makefile2:1334: recipe for target 'src/libopenrave/CMakeFiles/libopenrave.dir/all' failed
> make[1]: *** [src/libopenrave/CMakeFiles/libopenrave.dir/all] Error 2
> Makefile:148: recipe for target 'all' failed
> make: *** [all] Error 2
> ```

How did you fix this issue? I have the same problem. It doesn't recognize CopyFrom as a function. Ubuntu 20.04

cielavenir, 2022-05-14:
you would need rapidjson:master

(yes I agree that they should release a tagged version)