rdiankov/openrave #707: error: ‘rapidjson’ has not been declared

N3k4rk, 2019-10-21:
When building from commit 92cdaca62f3ce2e0a81140df28279b24c84d5e52
I get:

```
In file included from /home/openrave/git/openrave/include/openrave/openrave.h:2412:0,
                 from /home/openrave/git/openrave/src/boost_assertion_failed.cpp:6:
/home/openrave/git/openrave/include/openrave/planner.h:497:21: error: ‘rapidjson’ has not been declared
     void SaveToJson(rapidjson::Value& rPlannerStatus, rapidjson::Document::AllocatorType& alloc) const;
                     ^~~~~~~~~
/home/openrave/git/openrave/include/openrave/planner.h:497:37: error: expected ‘,’ or ‘...’ before ‘&’ token
     void SaveToJson(rapidjson::Value& rPlannerStatus, rapidjson::Document::AllocatorType& alloc) const;
                                     ^
src/CMakeFiles/boost_assertion_failed.dir/build.make:62: recipe for target 'src/CMakeFiles/boost_assertion_failed.dir/boost_assertion_failed.cpp.o' failed
make[2]: *** [src/CMakeFiles/boost_assertion_failed.dir/boost_assertion_failed.cpp.o] Error 1
CMakeFiles/Makefile2:688: recipe for target 'src/CMakeFiles/boost_assertion_failed.dir/all' failed
make[1]: *** [src/CMakeFiles/boost_assertion_failed.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

cielavenir, 2019-10-21:
not sure we can remove rapidjson dependency from the current (heavily depend) state. simply `make install` Tencent/rapidjson, or stick with 951676167e443eec6b40163d4f5b68d0858b74ef for now

N3k4rk, 2019-10-21:
installing rapidjson from source helped. I have had already installed rapidjson before, using
`sudo apt-get install rapidjson-dev`
to fix 
`libopenrave-core/libopenrave0.10-core.so.0.10.0: undefined reference to `OpenRAVE::InterfaceBase::SupportsJSONCommand(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status`

cielavenir, 2019-10-21:
perhaps you would need to remove cmake cache and reconfigure

PeterBowman, 2019-11-04:
Looks like rapidjson is a hard (required) dependency on latest master. The following block as well as all occurrences of `OPENRAVE_RAPIDJSON` could be removed, I guess:

https://github.com/rdiankov/openrave/blob/abe85928e74381d3d3d33db73693d92c9fbd7a1c/CMakeLists.txt#L537-L543

N3k4rk, 2019-11-06:
Just in case somebody missed it due to how my comment was worded (i've updated it now): installing rapidjson from source on top of apt-get installation fixed the issue for me.

nxdefiant, 2019-12-31:
Installing rapidjson-dev on ubuntu 18.04 fixed this error me, no need to install from source. But I completely removed the cmake cache.