rdiankov/openrave #709: Cannot get image data from camera

achoe08, 2019-10-24:
Currently, I am trying to simulate an RGB camera in OpenRAVE, but when I configure the camera to power on, I receive this warning message:

2019-10-24 12:57:51,549 openrave [WARN] [qtcoinviewer.cpp:3452 _GetCameraImage] offscreen renderer failed (check video driver), disabling
2019-10-24 12:57:51,601 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,652 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,703 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,754 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,806 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,857 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,908 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:51,959 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,011 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,062 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,114 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,165 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,216 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,267 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,318 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,370 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,421 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,472 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,523 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,575 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,626 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,677 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,728 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,780 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,831 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,882 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,934 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:52,985 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,036 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,087 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,139 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,190 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,241 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,292 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,343 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,394 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,446 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,497 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,549 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,600 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,651 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,702 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,753 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,805 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,856 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,907 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen
2019-10-24 12:57:53,958 openrave [WARN] [qtcoinviewer.cpp:3405 _GetCameraImage] cannot render offscreen

I am currently running on Ubuntu 16.04 on my Mac through VMware Fusion. Any help would be appreciated

N3k4rk, 2019-10-29:
My guess would be its because of the virtual machine and its graphics driver. Im having issues with viewers in Ubuntu running in Virtualbox. Theoretically you could try to use qtosg instead of qtcoin as a viewer.

jgvictores, 2019-12-22:
Here are some guidelines. The bottom line from our research is that you do need NVIDIA drivers, and also an X11 file setting. Would be great to know if you get this working on a VM! http://robots.uc3m.es/gitbook-installation-guides/install-openrave.html#offscreen-rendering-openrave-rgb-cameras ([perma](https://github.com/roboticslab-uc3m/installation-guides/blob/0cf20e282437bde0a68d8fa75681c52ee5e2b365/install-openrave.md#offscreen-rendering-openrave-rgb-cameras)).

PS: No, it seems this is not implemented in `qtosg`, see comments starting at https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/48#issuecomment-566170030