rdiankov/openrave #717: ikfast.py failed to generate .cpp file for my ur10e.dae (ubuntu 18.04)

mustang66ytz, 2019-12-02:
I have tried using the ikfast.py to generate a cpp code for the ikfast plugin, but failed everytime.
Here is the command I am using on my Ubuntu 18.04 machine:
> python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot ur10e.dae --iktype=transform6d --baselink=1 --eelink=8 --savefile=/home/ericyang/ws_fmcg_test/src/ompl_test/urdf/ikfastur10e.cpp

Here is the error:

INFO: c=1482, store solution
INFO: c=1483 var=j4
INFO: c=1484, store solution
INFO: c=1485 var=j4
INFO: c=1486, store solution
INFO: c=1487 var=j4
INFO: c=1488, store solution
INFO: c=1489 var=j4
INFO: c=1490, store solution
INFO: c=1491 var=j4
INFO: c=1492, store solution
INFO: c=1493 var=j4
INFO: c=1494, store solution
INFO: c=1495 var=j4
INFO: c=1496, store solution
INFO: c=1497 var=j4
INFO: c=1498, store solution
INFO: c=1499 var=j4
INFO: c=1500, store solution
INFO: c=1501 var=j4
INFO: c=1502, store solution
INFO: c=1503 var=j4
INFO: c=1504, store solution
INFO: c=1505 var=j4
INFO: c=1506, store solution
INFO: c=1507 var=j4
INFO: c=1508, store solution
INFO: c=1509 var=j4
INFO: c=1510, store solution
INFO: c=1511 var=j4
INFO: c=1512, store solution
INFO: c=1513 var=j4
INFO: c=1514, store solution
INFO: c=1515 var=j4
INFO: c=1516, store solution
INFO: c=1517 var=j4
INFO: c=1518, store solution
INFO: c=1519 var=j4
INFO: c=1520, store solution
INFO: c=1521 var=j4
INFO: c=1522, store solution
INFO: c=1523 var=j4
INFO: c=1524, store solution
INFO: c=1525 var=j4
INFO: c=1526, store solution
INFO: c=1527 var=j4
INFO: c=1528, store solution
INFO: c=1529 var=j4
INFO: c=1530, store solution
INFO: c=1531 var=j4
INFO: c=1532, store solution
INFO: c=1533 var=j4
INFO: c=1534, store solution
INFO: c=1535 var=j4
INFO: c=1536, store solution
INFO: c=1537 var=j4
INFO: c=1538, store solution
INFO: c=1539 var=j4
INFO: c=1540, store solution
INFO: c=1541 var=j4
INFO: c=1542, store solution
INFO: c=1543 var=j4
INFO: c=1544, store solution
INFO: c=1545 var=j4
INFO: c=1546 var=j0
INFO: c=1547, store solution
INFO: c=1548 var=j4
INFO: c=1549 var=j0
INFO: c=1550, store solution
Segmentation fault (core dumped)

Thanks for any suggestions! Without this, I cannot proceed with the ik calculation in openRAVE.

jgvictores, 2020-01-25:
Try with OpenRAVE v0.9.0 and sympy version '0.7.1'. If it works like that, maybe it will be easier to track down what has caused a regression.

popcornchen, 2022-02-15:
Exactly the same problem... I think the generation of iksolver has succeeded, I don't know why I can't find them in the target path..