rdiankov/openrave #724: C as output language / build errors on old version?

rosslloydqbot, 2020-01-14:
Hi
I am using LinuxCNC and as a result need IKFast to produce C based files rather than C++. In order to do this I have been trying to use a customised version of OpenRave which I found here: https://github.com/ftkalcevic/openrave/tree/linuxcnc_gen_c . 

However it is based on a very old version of openrave and when building I get a lot of errors of the type:

```
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
plugins/qtcoinrave/CMakeFiles/qtcoinrave.dir/build.make:62: recipe for target 'plugins/qtcoinrave/moc_qtcoinviewer.cxx' failed
```

I found advice on stack overflow that this has to do with a boost / Qt conflict (I am using Qt version 5.5.1 and Boost 1.58) and to encapsulate all boost includes with:
```
#ifndef Q_MOC_RUN
#endif
```
As each error came up, I located the files it was building, and for any boost includes I added the above. However, I am still getting the error when trying to build either **pqrave.cpp**, **oderave.cpp** or **ik_pr2_head.cpp** (I cannot tell from the make terminal output) despite modifying their associated **plugindefs.h** . 

It would be better to just use an up to date and stable version of OpenRAVE, but I cannot tell if its possible to output as C rather than C++, at least from looking at the code here https://github.com/rdiankov/openrave/blob/master/python/databases/inversekinematics.py it does not seem possible.

Please can you advise either on how I might overcome the BOOST_JOIN make error or alternately if it is possible to output to C with modern OpenRAVE?
Thanks!

jgvictores, 2020-01-25:
Hi!

From https://github.com/ftkalcevic/openrave/tree/873b76c2f9629148300c2627b1b864f5e5668512 at time of writing we can see `This branch is 2 commits ahead, 2940 commits behind rdiankov:master.`

I'd propose a git rebase or cheery-pick on to `v0.9.0` or even `master` and see if you can get away with it (Boost 1.58 should be fine on the tag and branch I mention).

cielavenir, 2020-04-10:
@jgvictores I personally (merged it and) tried https://github.com/cielavenir/openrave/tree/linuxcnc_gen_c_master. But I just found that the generated iksolver C source file is somewhat broken and this part does not seem something I can fix.