rdiankov/openrave #732: databases/__init__.py 'robot.SetActiveManipulator()' Wrong?

zhubeilife, 2020-01-26:
It seems below code should 

```python
if hasattr(options,'manipname') and robot.IsRobot():
                    if options.manipname is None:
                        # prioritize manipulators with ik solvers
                        indices = [i for i,m in enumerate(robot.GetManipulators()) if m.GetIkSolver() is not None]
                        if len(indices) > 0:
                            robot.SetActiveManipulator(indices[0])
                    else:
                        for i,m in enumerate(robot.GetManipulators()):
                            if m.GetName()==options.manipname:
                                robot.SetActiveManipulator(m)
                                break
```

change 

```python
indices = [i for i,m in enumerate(robot.GetManipulators()) if m.GetIkSolver() is not None]
```

to

```python
indices = [m for i,m in enumerate(robot.GetManipulators()) if m.GetIkSolver() is not None]
```

zhubeilife, 2020-01-26:
@rdiankov

cielavenir, 2020-02-09:
I would like to have least code to confirm this code path.

```
import openravepy
openravepy.databases.DatabaseGenerator.RunFromParser(args=['barrettwam.robot.xml'])
```

was ok for this case. will soon make PR for me.

zhubeilife, 2020-02-09:
> I would like to have least code to confirm this code path.
> 
> ```
> import openravepy
> openravepy.databases.DatabaseGenerator.RunFromParser(args=['barrettwam.robot.xml'])
> ```
> 
> was ok for this case. will soon make PR for me.

Sorry I didn't get it? Do you need me to run the above code?

cielavenir, 2020-02-09:
> Do you need me to run the above code?

I needed but don't need anymore.

zhubeilife, 2020-02-10:
@cielavenir Thanks

closed 2020-02-10 by zhubeilife