rdiankov/openrave #753: Suitability of 5 axis robot for using IKFast

rosslloydqbot, 2020-03-04:
Hi

(cross posted from ROS Answers)

 Openrave version: https://github.com/rdiankov/openrave
   (master) built with advice from
   https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html

ROS version: kinetic

Linux Distro: Ubuntu 16.04

I am a newcomer to openrave and ikfast. I am working on a robot that is very similar to a 5 axis mill in design, but with some unusual offsets in the final two rotational axes. Here is a link to a simplified kinematic diagram (using pasteboard as I do not have enough karma to post the jpeg):

[Diagram](https://pasteboard.co/IXxzeZ6.jpg)

It is basically a PPPRR robot. I extracted a URDF from solidworks for our robot, and the URDF works well in ROS / RViz. I was able to build a moveit package and via KDL, was able to plan and move correctly.

However when using the latest stable version of IKFast from the github, I am repeatedly coming up against errors:

    python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot=saws_urdf.dae --iktype=translationdirection5d --baselink="0" --eelink="5" --savefile="$IKFAST_OUTPUT_PATH"
    INFO: moved translation [0, 0, 41/200] to right end
    INFO: moved translation [0, 0, 0] to left end
    INFO: moved translation on intersecting axis [6139/12500, -43143/100000, -40543/100000] to left
    INFO: [[0, 0, -1, 0],[0, 1, 0, 0],[1, 0, 0, 0]]
    INFO: [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, j0]]
    INFO: [[0, 0, 1, 0],[0, 1, 0, 0],[-1, 0, 0, 0]]
    INFO: [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, j1]]
    INFO: [[1, 0, 0, 6139/12500],[0, 0, 1, -40543/100000],[0, -1, 0, 43143/100000]]
    INFO: [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, j2]]
    INFO: [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, 0]]
    INFO: [[cos(j3), -sin(j3), 0, 0],[sin(j3), cos(j3), 0, 0],[0, 0, 1, 0]]
    INFO: [[0, 0, 1, 0],[1, 0, 0, 0],[0, 1, 0, -61/1000]]
    INFO: [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
    INFO: [[0, 0, -1, 0],[0, 1, 0, 0],[1, 0, 0, 41/200]]
    INFO: ikfast translation direction 5d: [j0, j1, j2, j3, j4], direction=[0]
    [0]
    [1]
    INFO: computed in 0.536197s
    INFO: attempting li/woernle/hiller general ik method
    WARNING: symbols [j2, cj3, sj3, cj4, sj4] are not all rotational, is this really necessary?
    WARNING: CannotSolveError: symbols [j2, cj3, sj3, cj4, sj4] are not all rotational, is this really necessary?
    INFO: computed in 0.415236s
    INFO: attempting li/woernle/hiller general ik method
    WARNING: CannotSolveError: Li/Woernle/Hiller method requires 3 unknown variables, has 2
    INFO: attempting kohli/osvatic general ik method
    WARNING: CannotSolveError: Kohli/Osvatic method requires 3 unknown variables
    INFO: attempting kohli/osvatic general ik method
    WARNING: CannotSolveError: Kohli/Osvatic method requires 3 unknown variables
    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 9521, in <module>
        chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
      File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2281, in generateIkSolver
        chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
      File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2803, in solveFullIK_TranslationDirection5D
        raise self.CannotSolveError('raghavan roth equations too complex')
    __main__.CannotSolveError: CannotSolveError: raghavan roth equations too complex

Here are the `--info links` and `--info joints`outputs:

    $ openrave-robot.py "$MYROBOT_NAME".dae --info links
    name      index parents  
    -------------------------
    base_link 0              
    xaxis     1     base_link
    yaxis     2     xaxis    
    zaxis     3     yaxis    
    caxis     4     zaxis    
    baxis     5     caxis    
    tool      6     baxis    
    -------------------------
    name      index parents 
    
    
    $ openrave-robot.py "$MYROBOT_NAME".dae --info joints
    name              joint_index dof_index parent_link child_link mimic
    --------------------------------------------------------------------
    xaxis_base_joint  0           0         base_link   xaxis           
    yaxis_axis_joint  1           1         xaxis       yaxis           
    zaxis_yaxis_joint 2           2         yaxis       zaxis           
    caxis_zaxis_joint 3           3         zaxis       caxis           
    baxis_caxis_joint 4           4         caxis       baxis           
    tool_joint        -1          -1        baxis       tool            
    --------------------------------------------------------------------

The tool joint is fixed. 

I have tried using the xml wrapper approach on the off-chance it would make any difference, and it did not. 

To the title of the post: Before going into any further troubleshooting, I read that IKFast relies on the final axes intersecting, and I am not sure if ours qualify, due to the end offsets. If so am I out of luck for using IKFast even though KDL works well? Unfortunately the end application is not in ROS, or I could just use KDL. Though the project will eventually move into ROS, I am only using ROS as a known-good test bed for the IKFast generator before using a special version of openrave designed to output c files for linuxcnc. (Fallback option would be to hand calc the IK and modify an existing LinuxCNC kinematics c module, but I want to leverage the collision aware aspect of IKFAST.)

I am reluctant to post URDF's and meshes as this is a project for a company, however if the robot theoretically should work and more troubleshooting is in fact needed, please let me know and I will post the URDF. 

Thank you

rosslloydqbot, 2020-03-09:
Hi, in the end I wrote a custom kinematics module instead. I will close the issue.

closed 2020-03-09 by rosslloydqbot

Nurullahdmrly, 2021-12-13:
@rosslloydqbot Hi,
I have same issue these days and I didn't solve this problem. How did you write own custom module? Were you able to use this module with moveit? I am waiting your response and helping. if you help me , I'ld appreciate it.