rdiankov/openrave #756: 【Ubuntu 19.04, Problem with GUI】

Merikanto, 2020-03-08:
So I managed to install openrave, and could run it in Ubuntu 19.04.

But I had trouble invoking the GUI when I tried to run an example.

    $ openrave.py --example graspplanning

And I got the following error:

    [plugindatabase.h:645 Create] Failed to create name ivmodelloader, interface module

This is the only error I got, and everything else worked fine for me.

I tried to google for solutions, but got no luck. Still stuck with this error.

Does anyone know how to fix this? Thanks a lot!

jgvictores, 2020-03-10:
It looks like `ivmodelloader` ([ref](http://openrave.org/docs/0.8.0/interface_types/module/ivmodelloader/)) requires `qtcoin`. Does simply running `openrave` open a viewer?

Merikanto, 2020-03-11:
Okay, but that didn't solve the problem...

```shell
$ openrave
[openrave.cpp:322 MainOpenRAVEThread] failed to find an OpenRAVE viewer.
```

So how could I install the dependencies? Thank you!

jgvictores, 2020-03-11:
On Ubuntu 18.04:
```
sudo apt install libsoqt4-dev libcoin80-dev
```
Then run CMake (e.g. `cmake ..` from your `build/`) again and recompile (`make;  sudo make install`).

The problem here is that `libsoqt4-dev` doesn't seem to exist on Ubuntu 19.04. You can try with `libsoqt520-dev`, but I'm really not sure about results.

PS: If you decide to downgrade (possibly via re-installation) to 18.04 you'll get the benefit of an LTS (still supported) distro. However, it is nice to try to advance towards more modern dependencies, so it's cool if you can give it a try.

Merikanto, 2020-03-11:
Hello, so I tried to install libsoqt4-dev and libcoin80-dev, but I got the following notice:

```shell
Package libcoin80-dev is not available,   ......
However the following packages replace it:
  libcoin-dev:i386 libcoin-dev

Package libsoqt4-dev is not available,    ......
However the following packages replace it:
  libsoqt520-dev:i386 libsoqt520-dev

```

According to the notice, the replacement packages should do the work, however the GUI still cannot be invoked.

jgvictores, 2020-03-11:
Mmmm could you try `sudo apt install libsoqt520-dev libcoin-dev`, clear your build, and copy the log of a fresh `cmake ..` here?

Merikanto, 2020-03-11:
So I reinstalled the two packages, and recompile everything, then got the following result:

```shell
$ openrave.py --example graspplanning
[plugindatabase.h:1014 _SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: libosgUtil.so.132: cannot open shared object file: No such file or directory
[plugindatabase.h:1014 _SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: libosgUtil.so.132: cannot open shared object file: No such file or directory
[plugindatabase.h:941 _LoadPlugin] failed to load: /usr/local/lib/openrave0.9-plugins/libqtosgrave.so
[plugindatabase.h:645 Create] Failed to create name ivmodelloader, interface module
```

jgvictores, 2020-03-11:
Could you copy/upload the following?
- Generated `CMakeCache.txt` file
- Log of the cmake command

Merikanto, 2020-03-11:
`cmake` commands:

```shell
cd build
cmake .. -DOPT_VIDEORECORDING=OFF
make -j$(nproc)
sudo make install
```

`CMakeCache.txt` file:
https://raw.githubusercontent.com/Merikanto/Files/master/CMakeCache.txt

jgvictores, 2020-03-12:
In your `CMakeCache.txt` we read `COIN_CONFIG_EXECUTABLE:FILEPATH=COIN_CONFIG_EXECUTABLE-NOTFOUND`, which should point to `/usr/bin/coin-config`.
While `coin-config` was present in `libcoin80-dev` of Xenial 18.04 ([ref](https://packages.ubuntu.com/search?suite=xenial&arch=any&searchon=contents&keywords=coin-config)), it seems it was left out at `libcoin-dev` of Disco 19.04 ([ref](https://packages.ubuntu.com/search?suite=disco&arch=any&searchon=contents&keywords=coin-config)).

To avoid downgrading distro... how about compiling from source? Here are the repos:
- https://github.com/coin3d/coin
- https://github.com/coin3d/soqt

Merikanto, 2020-03-12:
Good idea.... Thank you, I will choose to build from source then!

I hope this could work. :)

Sendagaya, 2020-03-13:
I built the two repos from source, then rebuilt openrave again, but this still doesn't solve the problem...

(BTW I just upgraded to 19.10 Eoan last night, but I don't think there are much changes compared to 19.04).

Here's the warning I got when running an openrave example:

```
openrave.py --example graspplanning

[plugindatabase.h:645 Create] Failed to create name ivmodelloader, interface module
```

It's the same warning as before. The new `CMakeCache.txt` file:
https://raw.githubusercontent.com/Merikanto/Files/master/NEW_CMakeCache.txt

It indicated that `coin-config` executable is found:

```
//coin-config executable
COIN_CONFIG_EXECUTABLE:FILEPATH=/usr/local/bin/coin-config
```

But still, some other plugins are missing... (A bunch of `NOTFOUND`)

I'm not sure what exactly is missing here....

Merikanto, 2020-03-13:
I built the two repos from source, then rebuilt openrave again, but this still doesn't solve the problem...

(BTW I just upgraded to 19.10 Eoan last night, but I don't think there are much changes compared to 19.04).

Here's the warning I got when running an openrave example:

openrave.py --example graspplanning

[plugindatabase.h:645 Create] Failed to create name ivmodelloader, interface module
It's the same warning as before. The new CMakeCache.txt file:
https://raw.githubusercontent.com/Merikanto/Files/master/NEW_CMakeCache.txt

It indicated that coin-config executable is found:

//coin-config executable
COIN_CONFIG_EXECUTABLE:FILEPATH=/usr/local/bin/coin-config
But still, some other plugins are missing... (A bunch of NOTFOUND)

I'm not sure what exactly is missing here....

jgvictores, 2020-03-15:
In your `CMakeCache.txt` we read `SOQT_CONFIG_EXECUTABLE:FILEPATH=SOQT_CONFIG_EXECUTABLE-NOTFOUND`, which should point to some `soqt-config` file.

Merikanto, 2020-03-15:
Got it, thank you! 😂😂
I'm not sure if this has anything to do with OpenScene, but I'll see...

jgvictores, 2020-03-15:
> I'm not sure if this has anything to do with OpenScene, but I'll see...

Even if you got OpenScene working I do not think it will help. From the original [ref](http://openrave.org/docs/0.8.0/interface_types/module/ivmodelloader/), the qtcoin plugin is the one that is required (and they are independent).

Merikanto, 2020-03-15:
I saw this issue here with Ubuntu 18.04:

https://github.com/roboticslab-uc3m/installation-guides/issues/65

I wonder if openrave works successfully with bionic now....
I'd try & use openrave with Docker if everything works fine with 18.04...

(I did build `soqt` from source, yet the path wasn't found, which is weird... Please let me know if you could successfully run openrave on Ubuntu 19.04 / 19.10......)

jgvictores, 2020-03-16:
Oh, if you go for Ubuntu 18.04 everything will be fine, just follow http://robots.uc3m.es/gitbook-installation-guides/install-openrave.html#install-openrave-ubuntu-1804 ([perma](https://github.com/roboticslab-uc3m/installation-guides/blob/31a01a0390b681f3ce4151d8cba1b8512ae3693c/install-openrave.md#install-openrave-ubuntu-1804)).

The TLDR: We left that issue open because we are not totally satisfied with the solution. At the link we point to a copy of a fork of OpenRAVE 0.9.0 which has been modified for more modern Boost, whereas the main OpenRAVE fork is already at least at 0.25 and still not prepared for that version of Boost the last time we checked.

(most burns down to finding the correct versions and procedure, the latest `soqt` provides no `soqt-config` but a instead `soqt-config.cmake` (at least when built via cmake, and have not been able to build it via autotools yet))

Merikanto, 2020-03-16:
Thank you for the info! 
Shall I close this thread or leave it open?

jgvictores, 2020-03-16:
Hi! If you're happy with that solution I guess you can close the thread.

As a TLDR for the future: the `qtcoin` viewer of OpenRAVE (which is required for lots of stuff, such as virtual cameras) at time of writing looks for `coin-config` and `soqt-config`. Compiling both from source, we see no `soqt-config` (which is kind of `Makefile`-ish), and instead find a `soqt-config.cmake`, so possibly the future could be a PR to transition towards a more `CMake`-ish approach.

Merikanto, 2020-03-16:
Thank you! 

I think I'll leave it open for a couple more days.... to find out the right configuration.....

closed 2020-03-25 by Merikanto

cielavenir, 2020-03-26:
Hi there,

I have established openrave build on Ubuntu Focal: https://github.com/cielavenir/mujin_recruiting/blob/pybind11/cookbooks/mujin/recipes/default.rb (maybe you don't need jenkins though)
**This one uses pybind11 as python binding, not boost-python.**

A movie where graspplanning is running: https://www.dropbox.com/s/pb6pql1mjowmcun/graspplanning_focal.mp4?dl=0

I have not checked about ivmodelloader, but qtosg looks working as is.

I have also checked qtosg availability in Ubuntu Bionic as well, so the script should also work on Eoan.

Merikanto, 2020-03-26:
Wow! Thank you!