rdiankov/openrave #782: Openrave wiki is down

sirop, 2020-04-06:
http://openrave.programmingvision.com/wiki/index.php/Main_Page yields:

> MediaWiki internal error.
> 
> Exception caught inside exception handler.
> 
> Set $wgShowExceptionDetails = true; at the bottom of LocalSettings.php to show detailed debugging information.

zhubeilife, 2020-04-07:
@rdiankov Same Issue, the wiki pages is down

rdiankov, 2020-04-07:
@cielavenir someone should probably take a look. can you talk to me tomorrow in person?

cielavenir, 2020-04-07:
oh...? I have little knowledge about mediawiki. May I ask who was handling the wiki previously?

Yu-Ru, 2020-07-27:
The wiki link is still dead. Is there any update of the repair? Many thanks.

zhubeilife, 2020-07-28:
for temp using, we can refer to some web archive

https://web.archive.org/web/20191022170834/http://openrave.programmingvision.com/wiki/index.php/Format:XML

Yu-Ru, 2020-07-28:
@zhubeilife Many thanks for the link. The resource there is very helpful.

Merikanto, 2021-01-26:
Hmm.... The openrave wiki is still down..... 😱😱