rdiankov/openrave #790: build error in ubuntu16.04 ,the lastest stable version.

wuxiangrui188, 2020-04-27:
i install the software cite this way: https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html
i have got clone the latest_stable .but when i run in cmmand "make - j4",it has something error as follows.i am a new for openrave.thanks a lot!

/home/wuxr/collada-dom/build/OpenSceneGraph/build/fcl/build/openrave/plugins/logging/viewerrecorder.cpp:876:40: error: ‘av_destruct_packet’ was not declared in this scope
                 av_destruct_packet(&pkt);
                                        ^
/home/wuxr/collada-dom/build/OpenSceneGraph/build/fcl/build/openrave/plugins/logging/viewerrecorder.cpp:880:32: error: ‘av_destruct_packet’ was not declared in this scope
         av_destruct_packet(&pkt);
                                ^
/home/wuxr/collada-dom/build/OpenSceneGraph/build/fcl/build/openrave/plugins/logging/viewerrecorder.cpp: In member function ‘void ViewerRecorder::_StartVideo(const string&, double, int, int, int, int)’:
/home/wuxr/collada-dom/build/OpenSceneGraph/build/fcl/build/openrave/plugins/logging/viewerrecorder.cpp:795:44: warning: ignoring return value of ‘int avformat_write_header(AVFormatContext*, AVDictionary**)’, declared with attribute warn_unused_result [-Wunused-result]
         avformat_write_header(_output,NULL);
                                            ^
plugins/logging/CMakeFiles/logging.dir/build.make:75: recipe for target 'plugins/logging/CMakeFiles/logging.dir/viewerrecorder.cpp.o' failed
make[2]: *** [plugins/logging/CMakeFiles/logging.dir/viewerrecorder.cpp.o] Error 1
CMakeFiles/Makefile2:1858: recipe for target 'plugins/logging/CMakeFiles/logging.dir/all' failed
make[1]: *** [plugins/logging/CMakeFiles/logging.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

liuxuyang2000, 2020-06-30:
I've had the same problem and I haven't solved it yet.

Texas-C, 2020-07-03:
hi~

It seems you missed the `libavdevice-dev`.
Maybe you could try those scritps: [openrave-installation](https://github.com/crigroup/openrave-installation).