rdiankov/openrave #802: pybind11 compilation fail, wrong version?

hungpham2511, 2020-05-14:
Hi guys,

I was trying to compile openrave with pybind11 bindings, looking to use the library with Python3, using `-DUSE_PYBIND11_PYTHON_BINDINGS:BOOL=TRUE .` flag.

Had the below error,

```sh
...openravepy/pybind11/pybind11_bindings.h:145:14: error: invalid conversion from ‘size_t {aka long unsigned int}’ to ‘const char*’ [-fpermissive]
         arr[i] = v[i];
```

apparently the code that breaks is using operator[] of py::array_t, which is not valid as of my copy of pybind11 (v2.5.0). I need to use `mutable_at` in place of the operator to get it compiling.

```c++
// std::vector<bool> is special
template <>
inline py::array_t<bool> toPyArray(const std::vector<bool>& v)
{
    py::array_t<bool> arr;
    arr.resize({(int) v.size()});
    for(size_t i = 0; i < v.size(); ++i) {
        arr[i] = v[i];
    }
    return arr;
}
```

Most likely you guys are using a different version of pybind11. If so, could you please point it out? Many thanks! Please also let me know if there is anything I can do to help/make the issue clearer.

lianghongzhuo, 2020-05-14:
```
git clone https://github.com/pybind/pybind11.git && mkdir pybind11/build
cd pybind11/build
git remote add woody https://github.com/woodychow/pybind11.git
git fetch woody
git checkout v2.2.4
git cherry-pick 94824d68a037d99253b92a5b260bb04907c42355
git cherry-pick 98c9f77e5481af4cbc7eb092e1866151461e3508
cmake .. -GNinja -DPYBIND11_TEST=OFF -DPythonLibsNew_FIND_VERSION=2
ninja -j4 && ninja install
```

Using this patch works, maybe @cielavenir you can explain a lit bit why your patch works while the master branch can't?

hungpham2511, 2020-05-15:
Awesome. Many thanks @lianghongzhuo 

I can see that this compile the bindings using Python2. Does that mean the pybind11 binding is not supposed to work with Python3.x?

lianghongzhuo, 2020-05-15:
It should also work in python 3, but I didn't check. this patch is a from @cielavenir for his project. Feel free to try compile without the python2 flag.

hungpham2511, 2020-05-15:
I can compile with the patch, though with Python2.7. Looking at the CMakeLists.txt it seems like OpenRAVE has not supported yet Python3.x. In any case, many thanks for the help. Closing the issue now.

closed 2020-05-15 by hungpham2511