rdiankov/openrave #808: Error compiling IKfast module on latest production branch

hungpham2511, 2020-05-20:
I am running the lastest production branch (as of a few days ago), on Ubuntu 16.04. Running `ikmodel.autogenerate` fail with the below error:

```
/home/hung/.openrave/kinematics.72948cfc3ff77d3858ae895ad25226f4/ikfast0x1000004b.IkParameterizationType.Transform6D.x86_64.15_16_18_19_20_21_f12_17.cpp:12404:11: error: ‘class ikfast::IkSingleDOFSolutionBase<double>’ has no member named ‘indices’
 vinfos[7].indices[0] = _ij21[0];
           ^
/home/hung/.openrave/kinematics.72948cfc3ff77d3858ae895ad25226f4/ikfast0x1000004b.IkParameterizationType.Transform6D.x86_64.15_16_18_19_20_21_f12_17.cpp:12405:11: error: ‘class ikfast::IkSingleDOFSolutionBase<double>’ has no member named ‘indices’
 vinfos[7].indices[1] = _ij21[1];
           ^
```

Look like there is a bug in the generation of the cpp file.

hungpham2511, 2020-05-20:
Any hints/pointers to the relevant source files are appreciated. Thanks!

tgn3000, 2020-05-21:
@hungpham2511 Hi, I wonder whether you could share the scene file (*.dae or *.xml) so that I try reproducing.

Can you also find the `ikfast.h` that you installed openrave with? See
```
template <typename T>
class IkSingleDOFSolutionBase {
...
    std::array<unsigned char, IKSINGLEDOFSOLUTIONBASE_INDICES_SIZE> indices; ///< unique index of the solution used to keep track on what part it came from. sometimes a solution can be repeated for different indices. store at least another repeated root
...
};
```

The `indices` member has been in this class for some years. Wondering which version did you upgrade from? Thanks!

hungpham2511, 2020-05-21:
Thanks for answering! I have a closer look at the compilation error and it turns out the compiler was looking for std=c++11. 
```
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
```
This is probably because I compile OpenRAVE with C++11. I will close this issue as it's not an issue with the source code.

hungpham2511, 2020-05-21:
@tgn3000 I think it's good to mention that I need to compile with C++11 to have pybind11 bindings working using this cmake command:
```
cmake -DODE_USE_MULTITHREAD=ON -DOSG_DIR=/usr/local/lib64/ -DUSE_PYBIND11_PYTHON_BINDINGS:BOOL=TRUE ..
```
cmake complains that it can't compile the test source with pybind headers. Quite strange.

closed 2020-05-21 by hungpham2511

tgn3000, 2020-05-21:
@hungpham2511 Good to know, thanks!

I think we have forced the use to compile with C++11 a few versions ago, 
https://github.com/rdiankov/openrave/blob/0324267a61b561d71426d77de8951e9168b95b92/CMakeLists.txt#L19
so it's natural you compiled with `-std=c++11`. However, when compiling the ikfast's raw cpp solver (`ikfast0x1000004b.*.cpp`), I guess it shouldn't matter whether it's `-std=c++11` or not? Just curious.

hungpham2511, 2020-05-21:
Could it be my copy of gcc? Anyway this is the error I got when trying to compile the generated source file directly:
```
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)
eureka@thinkpad:~/.openrave/kinematics.72948cfc3ff77d3858ae895ad25226f4$ g++ ikfast0x1000004b.IkParameterizationType.Transform6D.15_16_18_19_20_21_f12_17.cpp
In file included from /usr/include/c++/5/array:35:0,
                 from ikfast.h:34,
                 from ikfast0x1000004b.IkParameterizationType.Transform6D.15_16_18_19_20_21_f12_17.cpp:22:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
 ```

Compile ok with -std=c++11
```
eureka@thinkpad:~/.openrave/kinematics.72948cfc3ff77d3858ae895ad25226f4$ g++ -std=c++11 ikfast0x1000004b.IkParameterizationType.Transform6D.15_16_18_19_20_21_f12_17.cpp
eureka@thinkpad:~/.openrave/kinematics.72948cfc3ff77d3858ae895ad25226f4$
eureka@thinkpad:~/.openrave/kinematics.72948cfc3ff77d3858ae895ad25226f4$ ./a.out

Usage: ./ik r00 r01 r02 t0 r10 r11 r12 t1 r20 r21 r22 t2 free0 ...

Returns the ik solutions given the transformation of the end effector specified by
a 3x3 rotation R (rXX), and a 3x1 translation (tX).
There are 2 free parameters that have to be specified.
```

hungpham2511, 2020-05-21:
Update: I can compile ikfast fine now with this patch. https://github.com/hungpham2511/openrave/commit/ea80493cc0c112869c3e3683b456a380c3ced4c1

tgn3000, 2020-05-21:
> Update: I can compile ikfast fine now with this patch. [hungpham2511@ea80493](https://github.com/hungpham2511/openrave/commit/ea80493cc0c112869c3e3683b456a380c3ced4c1)

I see! Thanks for sharing~ If convenient, can you give Rosen a pull request?

tgn3000, 2020-05-21:
> -DUSE_PYBIND11_PYTHON_BINDINGS:BOOL=TRUE

Yes, assign it `TRUE` either at
https://github.com/rdiankov/openrave/blob/0324267a61b561d71426d77de8951e9168b95b92/CMakeLists.txt#L864
or at `cmake` command. Wondering if you successfully built openravepy with python2 or python3? I find it interesting to start seeing people compile with python3.

hungpham2511, 2020-05-21:
I tried a few days ago to compile the code with python3 using this cmake command: 
```
cmake -DODE_USE_MULTITHREAD=ON -DOSG_DIR=/usr/local/lib64/ -DUSE_PYBIND11_PYTHON_BINDINGS:BOOL=TRUE -DPYBIND11_PYTHON_VERSION=3.7 ..
```
Which works for a "simple" pybind11 cmake-based project, however, openrave complains about a mismatch version of the interpreter because it explicitly looks for a python2 interpreter instead.https://github.com/rdiankov/openrave/blob/0324267a61b561d71426d77de8951e9168b95b92/CMakeLists.txt#L94

Would be great if you can share your steps for compiling with python3.