rdiankov/openrave #867: Compilation error: pybind11::dict has no member named 'keys'

hungpham2511, 2020-08-07:
Hi guys, I am trying to compile the latest production branch but have this error. I can compile just fine before the PR that introduces this line which is #814 by @bucketzxm 

It seems like openrave needs to be compiled with a modified pybind source now. Appreciate if anyone can point me to the commit/custom pybind change.

The relevant error message:
```
../python/bindings/openravepy_kinbody.cpp:180:59: error: ‘class pybind11::dict’ has no member named ‘keys’
  180 |         py::list keys = py::list(pyReadableInterfacesDict.keys());
 ```

Many thanks!

cielavenir, 2020-08-08:
Yes I just found it yesterday and I attempted this:

```
diff --git a/python/bindings/openravepy_kinbody.cpp b/python/bindings/openravepy_kinbody.cpp
index b643fc605..f3ca935ea 100644
--- a/python/bindings/openravepy_kinbody.cpp
+++ b/python/bindings/openravepy_kinbody.cpp
@@ -176,8 +176,11 @@ std::map<std::string, ReadablePtr> ExtractReadableInterfaces(object pyReadableIn
     py::dict pyReadableInterfacesDict = (py::dict)pyReadableInterfaces;
     std::map<std::string, ReadablePtr> mReadableInterfaces;
     try {
-
+#ifdef USE_PYBIND11_PYTHON_BINDINGS
+        py::list keys = py::list(pyReadableInterfacesDict);
+#else
         py::list keys = py::list(pyReadableInterfacesDict.keys());
+#endif
         size_t numkeys = len(keys);
         for(size_t iKey = 0; iKey < numkeys; iKey++) {
             std::string name = py::extract<std::string>(keys[iKey]);
```

However I'm not convinced. Is this correct @bucketzxm @tgn3000 ? (note: this patch is one of the patch series I mentioned in mattermost 2to3 thread)

rdiankov, 2020-08-09:
@bucketzxm @tgn3000 let's resolve this ASAP

hungpham2511, 2020-08-11:
Confirm that the PR works. Many thanks! 

I will close this issue.

closed 2020-08-11 by hungpham2511