rdiankov/openrave #906: make error

csufangyu, 2020-10-20:
hi I get a error when i try to make.the eroor is:  
_```
/usr/include/c++/7/cmath:1484:5: error: no type named ‘__type’ in ‘struct __gnu_cxx::__promote<std::complex<long double>, false>’
/home/robot203/robot_project/onpenrave_robot/openrave-boost-1.6x/3rdparty/fparser-4.5/extrasrc/fpaux.hh:237:24: error: no matching function for call to ‘hypot(const std::complex<long double>&, const std::complex<long double>&)’
     { return std::hypot(x,y); }
              ~~~~~~~~~~^~~~~
```_
it maybe the version of c++?or others?
could you tell me how to solve it?thanks very much!

ValerioMa, 2020-10-22:
Same error here

> hi I get a error when i try to make.the eroor is:
>
> ```
> /usr/include/c++/7/cmath:1484:5: error: no type named ‘__type’ in ‘struct __gnu_cxx::__promote<std::complex<long double>, false>’ [...]

The same happens to me

csufangyu, 2020-10-22:
> Same error here
>
>
> The same happens to me

have you sloved it? I try to gcc-4.8,gcc-5.4,gcc-6,gcc-7,gcc-8,gcc-9,but they are no helpful!

csufangyu, 2020-10-25:
> Same error here
>
>
> The same happens to me

hi I have solved it.maybe you can reference this repositorie：
https://github.com/csufangyu/openrave-installation.git

closed 2020-10-25 by csufangyu

cielavenir, 2020-10-25:
I just noticed your fparser version is old - master branch has 3rdparty/fparser-4.5.2. And it **seems** that you used https://github.com/rdiankov/openrave/pull/625 according to the path "openrave-boost-1.6x/3rdparty/fparser-4.5". That is not compatible (even my branch https://github.com/cielavenir/openrave/tree/boost-1.6x-forcompile uses fparser-4.5.2).

In this way it was correct that you used that guide. (next time reporting git hash (or branch) will be helpful)