rdiankov/openrave #930: make fails with "array must be initialized with a brace-enclosed initializer" (master, Ubuntu 16.04)

atenpas, 2020-12-16:
Running `make` for the **master** branch on **Ubuntu 16.04** fails with errors like:

> include/openrave/kinbody.h:1143:75:` error: array must be initialized with a brace-enclosed initializer

and

> include/openrave/kinbody.h:1143:75: error: too many initializers for ‘boost::array<short int, 3ul>’

I suspect that this has to do with the **boost** library version?

cielavenir, 2020-12-16:
I have not taken detailed look,but please try `-DCMAKE_CXX_FLAGS=-std=gnu++11` or gnu++14 in cmake. (note: g++-6 defaults to gnu++14)

huiwenzhang, 2021-01-09:
Hi @atenpas , I fixed this by initializing all boost::array object with command `boost:array<T, size> object = {{ }}`, i.e. using double brace.