rdiankov/openrave #934: Does ikfast (type=transform6d) only work if there exist 3 consecutive intersecting axis?

junhyeokahn, 2020-12-28:
Hello, I am trying to generate a cpp file using ikfast for humanoid legs that have 6 DOF.
I was basically following [this instructure](http://docs.ros.org/en/kinetic/api/framefab_irb6600_support/html/doc/ikfast_tutorial.html#irb2400-test-links-info) and using a docker image from ```personalrobotics/ros-openrave```.
I have tried with a couple of different robots (i.e., Valkyrie, Atlas, and Nao), but I have been able to generate a file successfully only with the Nao robot.
I have closely looked at the urdfs of three robots and realize only Nao robot has three intersecting axes at the base whereas the other two does not have.
Indeed, when I modify the urdf of Atlas so that hip joints are located at the same position, it works.
I am wondering if it is true that for 6 DOF manipulator has to have three intersecting axes at the base to solve tranform6d type ik.
I am attaching three urdf files I used.
Any help or suggestions are appreciated.

[urdfs.zip](https://github.com/rdiankov/openrave/files/5745838/urdfs.zip)

tgn3000, 2021-01-06:
yes, it's true

junhyeokahn, 2021-01-08:
Thanks for the comment!

closed 2021-01-08 by junhyeokahn

frankymacster, 2022-01-28:
@tgn3000 why is it true?
Does it mean that the IK problem for robot arms such as the [CRX-10ia](https://www.fanuc.eu/es/en/robots/robot-filter-page/collaborative-robots/crx-10ia) can't be solved using IKFast?

Thank you