rdiankov/openrave #940: Problem with Installing OpenRAVE on Ubuntu 20.04

Merikanto, 2021-01-25:
Here's the error:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named sympy
-- failed to find python sympy system installation
-- System sympy (v=) is not right version, using local sympy
CMake Error at python/bindings/CMakeLists.txt:26 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/root/git/openrave/build/CMakeFiles/CMakeOutput.log".
See also "/root/git/openrave/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
```

**Note:**
I did have sympy installed, and I'm not sure why pybind11 cannot be found.
My Linus version is Ubuntu 20.04, and I followed this link: http://robots.uc3m.es/gitbook-installation-guides/install-openrave.html
I read through the troubleshooting part, but didn't find an issue that's similar to mine.

Any help would be appreciated! Thanks so much!

cielavenir, 2021-01-26:
I feel wrong python version is called; can you try https://github.com/cielavenir/openrave/commit/03d085f51e3db5b94a1049f09fdfd0c0a981fb42 ?
(if this works, I can suggest merging this to mainstream. currently this commit is just test)

Merikanto, 2021-01-26:
Thanks for your reply! Let me try...

cielavenir, 2021-01-26:
make sure to remove CMake cache before trying

Merikanto, 2021-01-26:
Got it!

Merikanto, 2021-01-26:
Hmm....

So I did remove CMakeCache:  ```rm openrave/build/CMakeCache.txt```

Then edited the line:  ```vi openrave/CMakeLists.txt```

Edits:
```python
  # include(FindPythonInterp)
  find_package(PythonInterp 2)
```

However, after I edited CMakeLists.txt and run ```sudo ./install-openrave.sh```, I got exactly the same error.

Then I went back and check the CMakeLists file, and realized that the I edited was overwritten to the original ```include(FindPythonInterp)``` again.

I tried several times, even reinstall everything, but got the same result. Btw my system's default python path is python 2.7.18 (```python --version```), so I really don't know what went wrong....

Merikanto, 2021-01-26:
Could it be that I don't have pip installed for python2.7?

```yaml
pip3 -V
> pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
```
I saw that pip for python2 is deprecated, running ```sudo apt install python-pip``` no longer works on Ubuntu 20.04,
and when I tried to install using this:
```shell
curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py
sudo python2 get-pip.py
```
It gives me error:
```shell
Traceback (most recent call last):
  File "get-pip.py", line 24226, in <module>
    main()
  File "get-pip.py", line 199, in main
    bootstrap(tmpdir=tmpdir)
  File "get-pip.py", line 82, in bootstrap
    from pip._internal.cli.main import main as pip_entry_point
  File "/tmp/tmpKhAs74/pip.zip/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax

```

cielavenir, 2021-01-26:
you need to remove `git reset --hard ${RAVE_COMMIT}` from install-openrave.bash

> Hmm....
> 
> So I did remove CMakeCache: `rm openrave/build/CMakeCache.txt`
> 
> Then edited the line: `vi openrave/CMakeLists.txt`
> 
> Edits:
> 
> ```python
>   # include(FindPythonInterp)
>   find_package(PythonInterp 2)
> ```
> 
> However, after I edited CMakeLists.txt and run `sudo ./install-openrave.sh`, I got exactly the same error.
> 
> Then I went back and check the CMakeLists file, and realized that the edited line was overwritten to the original `include(FindPythonInterp)` again.
> 
> I tried several times, even reinstall everything, but got the same result. Btw my system's default python path is python 2.7.18 (`python --version`), so I really don't know what went wrong....

cielavenir, 2021-01-26:
- install python-setuptools
- run `python -m easy_install 'pip<21'`

though I don't think this is the cause.

> Could it be that I don't have pip installed for python2.7?
>

Merikanto, 2021-01-26:
Got it!

So now the sympy version is resolved after pip for py2.7 is installed,
but pybind is still giving me errors (after trying ```git reset --hard ${RAVE_COMMIT}```)

```
-- Found sympy version '0.7.1'
-- Assuming sympy version 0.7.x or greater
CMake Error at python/bindings/CMakeLists.txt:26 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/root/git/openrave/build/CMakeFiles/CMakeOutput.log".
See also "/root/git/openrave/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
```

I wonder if other people encountered issues like this before.....

Merikanto, 2021-01-28:
Hey I checked pybind11, and I somehow think it's pybind's issue:

https://github.com/pybind/pybind11/issues/1379

pip install does not have the cmake file and other config files.

cielavenir, 2021-01-29:
you need to follow install-dependencies.sh's pybind11 installation... (openrave requires special branch as well)

Merikanto, 2021-01-29:
> you need to follow install-dependencies.sh's pybind11 installation... (openrave requires special branch as well)

Well, I did run ```./install-dependencies.sh``` at first.... And that doesn't work....
I actually tried to install pybind11 with anaconda, and that worked....

But my Ubuntu is running on a VM, and now the VM cannot be started.... (Blank screen at restart), so I cannot verify if it REALLY worked...

43998213, 2021-03-04:
> > you need to follow install-dependencies.sh's pybind11 installation... (openrave requires special branch as well)
> 
> Well, I did run `./install-dependencies.sh` at first.... And that doesn't work....
> I actually tried to install pybind11 with anaconda, and that worked....
> 
> But my Ubuntu is running on a VM, and now the VM cannot be started.... (Blank screen at restart), so I cannot verify if it REALLY worked...
> 
> **Update:**
> I tried again on a new VM, and that didn't work, still got the error from pybind11... :(

Is it working？

Merikanto, 2021-03-22:
> Is it working？

Hey it's not.... I have to use an old version of Ubuntu (16.04) to make it work...

cielavenir, 2021-03-23:
@Merikanto I full-tested openrave-installation; please try https://github.com/cielavenir/openrave-installation/tree/patch-1

Merikanto, 2021-03-23:
> @Merikanto I full-tested openrave-installation; please try https://github.com/cielavenir/openrave-installation/tree/patch-1

Thank you so much, will do!!

43998213, 2021-03-23:
> Hey it's not.... I have to use an old version of Ubuntu (16.04) to make it work...

Hey I try many time `./install-dependencies.sh` ,it's working on ubuntu20.04 . It is best to perform each step manually.

Thank you very much to the author.

#!/bin/bash
#
# Authors:
#   Francisco Suarez <fsuarez6.github.io>
#
# Description:
#   OpenRAVE Installation Script: Dependencies

# Check ubuntu version
UBUNTU_VER=$(lsb_release -sr)
if [ ${UBUNTU_VER} != '14.04' ] && [ ${UBUNTU_VER} != '16.04' ] && [ ${UBUNTU_VER} != '18.04' ] \
  && [ ${UBUNTU_VER} != '20.04' ]; then
    echo "ERROR: Unsupported Ubuntu version: ${UBUNTU_VER}"
    echo "  Supported versions are: 14.04, 16.04, 18.04, and 20.04"
    exit 1
fi

# Install dependencies
echo ""
echo "Installing OpenRAVE dependencies..."
echo ""
echo "Requires root privileges:"

# Update
if [ ${UBUNTU_VER} = '14.04' ]; then
  # ROS Indigo repository
  sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
  sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
  # Additional PPAs
  sudo apt-add-repository -y ppa:imnmfotmal/libccd
fi
sudo apt-get update

# Programs
sudo apt-get install -y --no-install-recommends build-essential cmake doxygen \
  g++ git octave python-dev python-setuptools wget mlocate
if [ ${UBUNTU_VER} = '14.04' ] || [ ${UBUNTU_VER} = '16.04' ] || [ ${UBUNTU_VER} = '18.04' ]; then
  sudo apt-get install -y --no-install-recommends ipython python-h5py python-numpy \
    python-pip python-wheel python-scipy
elif [ ${UBUNTU_VER} = '20.04' ]; then
  sudo apt-get install -y --no-install-recommends python2 curl
  curl https://bootstrap.pypa.io/get-pip.py --output get-pip.py
  python2 get-pip.py
  pip install ipython h5py numpy scipy wheel 
fi
if [ ${UBUNTU_VER} = '14.04' ]; then
  sudo apt-get install -y --no-install-recommends qt4-dev-tools zlib-bin
elif [ ${UBUNTU_VER} = '16.04' ] || [ ${UBUNTU_VER} = '18.04' ] || [ ${UBUNTU_VER} = '20.04' ]; then
  sudo apt-get install -y --no-install-recommends qt5-default minizip
fi

# Libraries
sudo apt-get install -y --no-install-recommends ann-tools libann-dev            \
libassimp-dev libavcodec-dev libavformat-dev libeigen3-dev libfaac-dev          \
libflann-dev libfreetype6-dev liblapack-dev libglew-dev libgsm1-dev             \
libmpfi-dev  libmpfr-dev liboctave-dev libode-dev libogg-dev libpcre3-dev       \
libqhull-dev libswscale-dev libtinyxml-dev libvorbis-dev libx264-dev            \
libxml2-dev libxvidcore-dev libbz2-dev
if [ ${UBUNTU_VER} = '14.04' ] || [ ${UBUNTU_VER} = '16.04' ] || [ ${UBUNTU_VER} = '18.04' ]; then
  sudo apt-get install -y --no-install-recommends libsoqt-dev-common libsoqt4-dev
elif [ ${UBUNTU_VER} = '20.04' ]; then
  sudo apt-get install -y --no-install-recommends libsoqt520-dev
fi
if [ ${UBUNTU_VER} = '14.04' ]; then
  sudo apt-get install -y --no-install-recommends collada-dom-dev libccd      \
  libpcrecpp0 liblog4cxx10-dev libqt4-dev
elif [ ${UBUNTU_VER} = '16.04' ] || [ ${UBUNTU_VER} = '18.04' ] || [ ${UBUNTU_VER} = '20.04' ]; then
  sudo apt-get install -y --no-install-recommends libccd-dev                  \
  libcollada-dom2.4-dp-dev liblog4cxx-dev libminizip-dev octomap-tools
fi

# Install boost
if [ ${UBUNTU_VER} = '14.04' ] || [ ${UBUNTU_VER} = '16.04' ] || [ ${UBUNTU_VER} = '20.04' ]; then
    sudo apt-get install -y --no-install-recommends libboost-all-dev libboost-python-dev
elif [ ${UBUNTU_VER} = '18.04' ]; then
    # Install boost 1.58 from source
    BOOST_SRC_DIR=~/git/boost_1_58_0
    mkdir -p ~/git; cd ~/git
    wget http://sourceforge.net/projects/boost/files/boost/1.58.0/boost_1_58_0.tar.gz -O ${BOOST_SRC_DIR}.tar.gz
    tar -xzf ${BOOST_SRC_DIR}.tar.gz
    cd ${BOOST_SRC_DIR}
    ./bootstrap.sh --exec-prefix=/usr/local
    ./b2 -j `nproc`
    sudo ./b2 -j `nproc` install threading=multi
fi

if [ ${UBUNTU_VER} = '20.04' ]; then
  # Install RapidJSON
  mkdir -p ~/git 
  cd ~/git && git clone https://github.com/Tencent/rapidjson.git
  cd rapidjson && mkdir build && cd build
  cmake .. && make -j `nproc` && sudo make install

  # Install Pybind
  cd ~/git && git clone https://github.com/pybind/pybind11.git 
  cd pybind11 && mkdir build && cd build 
  git remote add woody https://github.com/woodychow/pybind11.git \
    && git fetch woody && git checkout v2.2.4 \
    && git cherry-pick 94824d68a037d99253b92a5b260bb04907c42355 \
    && git cherry-pick 98c9f77e5481af4cbc7eb092e1866151461e3508 \
    && cmake .. -DPYBIND11_TEST=OFF -DPythonLibsNew_FIND_VERSION=2 \
    && sudo make install
fi

# updatedb for debugging purposes
sudo updatedb

closed 2021-11-08 by cielavenir