rdiankov/openrave #963: Incompatibility in inversekinematics.py when compiled with pybind11

hungpham2511, 2021-03-15:
It seems like when openrave is compiled with pybind11, the following lines in inversekinematics.py break:

https://github.com/rdiankov/openrave/blob/ec02e7ceda3d08e778ad88a0fe298b715a4253bb/python/databases/inversekinematics.py#L1130-L1133

and 

https://github.com/rdiankov/openrave/blob/ec02e7ceda3d08e778ad88a0fe298b715a4253bb/python/databases/inversekinematics.py#L1118

The reason is that enum in pybind11 does not exposed `.values` and `.name` methods.

hungpham2511, 2021-03-15:
It seems like pybind11 enum also produces a different string representation of the enum value: https://github.com/rdiankov/openrave/blob/ec02e7ceda3d08e778ad88a0fe298b715a4253bb/python/databases/inversekinematics.py#L592

cielavenir, 2021-03-16:
for `str(self.iktype) and `iktype.name`, https://github.com/cielavenir/pybind11/commit/dae2d434bd806eac67e38f3c49cfc91f46e4fd88 would be helpful.

I will take a look at `IkParameterizationType.values` when I have time.

hungpham2511, 2021-03-17:
Cool. This fix should work. 

Look like the `.values.` should return the same as the `__members__` attribute.

cielavenir, 2021-03-17:
https://github.com/cielavenir/pybind11/commit/2e08ce9ba75f5a2d87a6f12e6ab657ac78444e8e should fix it

cielavenir, 2021-04-05:
if you still have issues with my patch, please let me know.

note that the patch is already integrated in https://github.com/cielavenir/mujin_recruiting/blob/pybind11/cookbooks/mujin/recipes/default.rb .

closed 2021-04-05 by cielavenir

hungpham2511, 2021-04-06:
Thanks! It does. 

Btw, I am just curious, will OpenRAVE provides support for python3 binding at any point in the future?

cielavenir, 2021-04-06:
see https://github.com/rdiankov/openrave/pull/979