rdiankov/openrave #1003: undefined symbol: _ZN5boost6system15system_categoryEv

benthebear93, 2021-06-30:
#678 Similar error but i got it from building openrave

Consolidate compiler generated dependencies of target testparabolicramp
[ 80%] Built target qtcoinrave
[ 86%] Built target ikfastsolvers
Consolidate compiler generated dependencies of target rplanners
[ 92%] Built target libopenrave-core
Consolidate compiler generated dependencies of target openrave
Consolidate compiler generated dependencies of target libopenrave-core_c
[ 94%] Linking CXX executable openrave0.62
[ 94%] Built target testparabolicramp
[ 94%] Built target libopenrave-core_c
[ 99%] Built target rplanners
/usr/bin/ld: CMakeFiles/openrave.dir/openrave.cpp.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
//usr/local/lib/libboost_system.so.1.58.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:107: recipe for target 'src/openrave0.62' failed
make[2]: *** [src/openrave0.62] Error 1
CMakeFiles/Makefile2:1010: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

any reason it gives undefined reference to symbol '_ZN5boost6system15system_categoryEv'?

cielavenir, 2021-06-30:
@benthebear93 make sure your source tree contains https://github.com/rdiankov/openrave/pull/704

benthebear93, 2021-06-30:
@cielavenir 
 
I have managed  to solve this Issue during the make however similar error show up when i try this
openrave.py --example hanoi

i checked '#define BOOST_SYSTEM_NO_DEPRECATED' in the boost_assertion_failed file. 

openravepy is not set into PYTHONPATH env variable, attempting to add
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 52, in <module>
    import openravepy
  File "/usr/local/lib/python2.7/site-packages/openravepy/__init__.py", line 34, in <module>
    openravepy_currentversion = loadstable(__openravepy_version__)
  File "/usr/local/lib/python2.7/site-packages/openravepy/__init__.py", line 15, in loadstable
    return _loadversion('_openravepy_'+ver.replace('.', '_'))
  File "/usr/local/lib/python2.7/site-packages/openravepy/__init__.py", line 20, in _loadversion
    mainpackage = __import__("openravepy", globals(), locals(), [targetname])
  File "/usr/local/lib/python2.7/site-packages/openravepy/_openravepy_0_62/__init__.py", line 29, in <module>
    from .openravepy_int import *
ImportError: /usr/local/lib/python2.7/site-packages/openravepy/_openravepy_0_62/openravepy_int.so: undefined symbol: _ZN5boost6python6detail11init_moduleEPKcPFvvE

my env is 18.04 and i am using ROS melodic.
simple 'openrave' command works but not examples.

ldd /usr/local/lib/python2.7/site-packages/openravepy/_openravepy_0_62/openravepy_int.so
	linux-vdso.so.1 (0x00007fffe77eb000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe9973c2000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007fe996e45000)
	libboost_python.so.1.58.0 => /usr/local/lib/libboost_python.so.1.58.0 (0x00007fe996bfc000)
	libboost_thread.so.1.58.0 => /usr/local/lib/libboost_thread.so.1.58.0 (0x00007fe9969d5000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe9967cd000)
	libopenrave0.62-core.so.0 => /usr/local/lib/libopenrave0.62-core.so.0 (0x00007fe98dd90000)
	libopenrave0.62.so.0 => /usr/local/lib/libopenrave0.62.so.0 (0x00007fe98d498000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe98d0fa000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe98cee2000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe98caf1000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fe9994c3000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fe98c8d4000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe98c6b5000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe98c4b1000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fe98c2ae000)
	libboost_system.so.1.58.0 => /usr/local/lib/libboost_system.so.1.58.0 (0x00007fe98c0aa000)
	libxml2.so.2 => /usr/lib/x86_64-linux-gnu/libxml2.so.2 (0x00007fe98bce9000)
	libboost_filesystem.so.1.58.0 => /usr/local/lib/libboost_filesystem.so.1.58.0 (0x00007fe98bad1000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007fe98b708000)
	libpcrecpp.so.0 => /usr/lib/x86_64-linux-gnu/libpcrecpp.so.0 (0x00007fe98b4ff000)
	libcollada-dom2.4-dp.so.0 => /usr/lib/libcollada-dom2.4-dp.so.0 (0x00007fe98a9c0000)
	libassimp.so.4 => /usr/lib/x86_64-linux-gnu/libassimp.so.4 (0x00007fe989ff5000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007fe989c3d000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fe989a17000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007fe9897e2000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007fe9895b7000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fe989345000)
	libminizip.so.1 => /usr/lib/x86_64-linux-gnu/libminizip.so.1 (0x00007fe98913a000)
	liburiparser.so.1 => /usr/lib/x86_64-linux-gnu/liburiparser.so.1 (0x00007fe988f1f000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007fe988d05000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007fe988b00000)
	libicudata.so.60 => /usr/lib/x86_64-linux-gnu/libicudata.so.60 (0x00007fe986f57000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007fe986d50000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007fe986b18000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fe9868e6000)

and this is ldd /usr/local/lib/python2.7/site-packages/openravepy/_openravepy_0_62/openravepy_int.so

what could be causing this..?

cielavenir, 2021-07-01:
is /usr/local/lib/libboost_python.so.1.58.0 py2 or py3?

maybe ldd /usr/local/lib/libboost_python.so.1.58.0

benthebear93, 2021-07-01:
@cielavenir 

ldd /usr/local/lib/libboost_python.so.1.58.0
	linux-vdso.so.1 (0x00007fff9856a000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fa3d9086000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa3d8e6e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa3d8c4f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa3d885e000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa3d84c0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fa3d9658000)

those it suppose to have python2?
i have ROS which run by python2 and also anaconda(python2, python3)
i did all the installation related with openrave with python2 env in anaconda.
maybe mixed python3 and python2 causing the problem?

cielavenir, 2021-07-01:
umm `nm -D /usr/local/lib/libboost_python.so.1.58.0|grep init_module|c++filt`

let's see if it shows `PyModuleDef&` or `const char *`

benthebear93, 2021-07-01:
@cielavenir 
it shows 
0000000000031cf0 T boost::python::detail::init_module(PyModuleDef&, void (*)())

cielavenir, 2021-07-01:
So /usr/local/lib/libboost_python.so.1.58.0 is python3 version!

Currently openrave assumes python2. You need to locate python2 version of libboost_python.so.

benthebear93, 2021-07-01:
Im not sure how to change to python2.  I will try to figure out!

dal88236, 2021-10-18:
Hi @benthebear93 did you solve this issue?

benthebear93, 2021-10-19:
@dal88236 

I was not able to solve the issue. I only need to install openrave for ROS IKfast plugin, so i just used different computure which didn't have anaconda installed.

cielavenir, 2021-10-19:
I just remember that Ubuntu 18.04 requires pybind11 build

refer https://github.com/cielavenir/mujin_recruiting/blob/pybind11/cookbooks/mujin/recipes/default.rb for procedure