rdiankov/openrave #1043: shoulder-elbow-wrist configurations

frankymacster, 2022-01-25:
Hi,
I'm wondering if it's possible from the set of solutions of an IKFast generated IK solver to extract a specific solution with a given shoulder-elbow-wrist configuration.

For example, some robot simulation applications allow to choose a specific shoulder-elbow-wrist configuration:
- shoulder: front/rear
- elbow: up/down
- wrist: flip/non-flip
[image]

Thanks!

cielavenir, 2022-01-26:
@frankymacster If you specify ikreturn=True to FindIKSolutions(), special class instance is returned and extra info can be accessed via GetMapDataDict(). Please see solutionindices field.

frankymacster, 2022-01-26:
Thanks @cielavenir!
Will I be able to select a specific configuration from the IK solutions list?
For example if I only want the IK solution with the following configuration:
- shoulder: Front (F)
- elbow: Up (U)
- wrist: Non-flip (N)

can I access the value returned by `GetMapDataDict()` some way like `mapDataDict["FUN"]`?

cielavenir, 2022-01-26:
`GetMapDataDict()['solutionindices']`, mentioned as `_mapdata["solutionindices"]` in plugins/ikfastsolvers/ikfastsolver.cpp

frankymacster, 2022-01-26:
ok yes I see it [here](https://github.com/rdiankov/openrave/blob/748891e955b01c660afd4d9e869ded32f03b2205/plugins/ikfastsolvers/ikfastsolver.cpp#L1526)

but what do the `solutionindices` consist of? 
from that code it seems to be a `std::vector` of `dReal` (`double`).
How can I know which index corresponds to the following configuration?
- shoulder: Front (F)
- elbow: Up (U)
- wrist: Non-flip (N)

cielavenir, 2022-01-26:
the type is double but it actually represents 0-7 integers (for 6dof robots). that is the index for each solution. you have to figure out which one corresponds to shoulder-front_elbow-up_wrist-non-flip on your own.

frankymacster, 2022-01-26:
Ok I see, so `solutionindices` holds a list of 8 integers (for 6dof robots), one for each solution. Will the order of the solutions always be the same every time the IK solver is run on a different input?

For example, will the index corresponding to "shoulder-front_elbow-up_wrist-non-flip" always be the same in the `solutionindices` for a given robot on any rotation, position input?

cielavenir, 2022-01-26:
Sorry I'm not sure for this matter, https://github.com/rdiankov/openrave/pull/809 might be helpful...

frankymacster, 2022-01-26:
Thanks a lot @cielavenir! I will verify if the index corresponding to "shoulder-front_elbow-up_wrist-non-flip" always be the same in the solutionindices for a given robot on any rotation, position input and take a look #809.