rdiankov/openrave #1073: TypeError: cannot determine truth value of Relational

yushuiqiang, 2022-04-02:
PS D:\mingw64lib\OpenRAVE\bin> python `D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py --robot=robots/barrettwam.robot.xml --baselink=0 --eelink=7 --savefile=ik.cpp --freeindex=2

Traceback (most recent call last):
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 9631, in <module>
    chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 2332, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 2927, in solveFullIK_6D
    tree = self.TestIntersectingAxes(solvejointvars,Links, LinksInv,endbranchtree)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 3015, in TestIntersectingAxes
    return self.solve6DIntersectingAxes(T0links,T1links,transvars,rotvars,solveRotationFirst=solveRotationFirst, endbranchtree=endbranchtree)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 3219, in solve6DIntersectingAxes
    rottree += self.SolveAllEquations(AllEquations,curvars=currotvars,othersolvedvars=othersolvedvars,solsubs=self.freevarsubs[:],endbranchtree=storesolutiontree)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 6931, in SolveAllEquations
    rawsolutions=self.SolvePairVariables(raweqns,var0,var1,othersolvedvars,unknownvars=curvars+unknownvars)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 8935, in SolvePairVariables
    return self.SolvePairVariablesHalfAngle(raweqns,var0,var1,othersolvedvars)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 7806, in SolvePairVariablesHalfAngle
    possiblefinaleq = self.checkFinalEquation(Poly(peq.LC(),leftvar),subs)
  File "D:\mingw64lib\OpenRAVE\lib\site-packages\openravepy\_openravepy_/ikfast.py", line 8248, in checkFinalEquation
    if len(coeffs) == 1 and Abs(coeffs[0]) < 2*(10.0**-self.precision):
  File "C:\Python\Python37\lib\site-packages\sympy\core\relational.py", line 511, in __bool__
    raise TypeError("cannot determine truth value of Relational")
TypeError: cannot determine truth value of Relational

cielavenir, 2022-04-03:
Maybe you need sympy==0.7.1?

yushuiqiang, 2022-04-18:
PS C:\Users\86159> pip install sympy
Looking in indexes: https://mirrors.aliyun.com/pypi/simple
Requirement already satisfied: sympy in c:\python\python37\lib\site-packages (1.10.1)
Requirement already satisfied: mpmath>=0.19 in c:\python\python37\lib\site-packages (from sympy) (1.2.1)

cielavenir, 2024-06-14:
@yushuiqiang it should be fixed by https://github.com/rdiankov/openrave/pull/1391