rdiankov/openrave #1147: VERSION file at cr-libm interpreted as `<version>` header

PeterBowman, 2022-08-10:
To reproduce, clone latest OpenRAVE master (currently at https://github.com/rdiankov/openrave/commit/e0badbcfe664a7b7e55877019f3cc8aa9b2621b9) and configure with:

```cmake
cmake .. -DOPT_VIDEORECORDING=OFF -DOPT_PLUGINS=OFF -DOPT_PYTHON=OFF -DOPT_COLLADA=OFF -DOPT_CBINDINGS=OFF
```

Result:

```
[ 76%] Building CXX object src/CMakeFiles/boost_assertion_failed.dir/boost_assertion_failed.cpp.o
In file included from /usr/include/boost/config/detail/select_stdlib_config.hpp:24,
                 from /usr/include/boost/config.hpp:44,
                 from /usr/include/boost/config/workaround.hpp:41,
                 from /usr/include/boost/detail/workaround.hpp:8,
                 from /usr/include/boost/format.hpp:19,
                 from /home/bartek/wingit/openrave/src/boost_assertion_failed.cpp:7:
/home/bartek/wingit/openrave/3rdparty/crlibm-1.0beta4/version:1:1: error: expected unqualified-id before numeric constant
    1 | 1.0beta4
      | ^~~~~~~~
```

Here, Boost is trying to find a [`<version>` header](https://en.cppreference.com/w/cpp/header/version) (introduced in C++20), then fall back to `<cstddef>` if missing (see [select_stdlib_config.hpp](https://github.com/boostorg/config/blob/boost-1.74.0/include/boost/config/detail/select_stdlib_config.hpp#L14-L32) or the below code excerpt). It appears to pick the plain-text file [3rdparty/crlibm-1.0beta4/VERSION](https://github.com/rdiankov/openrave/blob/e0badbcfe664a7b7e55877019f3cc8aa9b2621b9/3rdparty/crlibm-1.0beta4/VERSION) (which contains the offending "1.0beta4" string) instead of any other system-installed header.

```cxx
// from boost/config/detail/select_stdlib_config.hpp at Boost.config 1.74.0

#if defined(__cplusplus) && defined(__has_include)
#  if __has_include(<version>)
#    include <version>
#  else
#    include <cstddef>
#  endif
#elif defined(__cplusplus)
#  include <cstddef>
#else
#  include <stddef.h>
#endif
```

Environment:
* Ubuntu 22.04.1 (Jammy)
* CMake 3.22.1
* Compiler: GNU 11.2.0
* Boost 1.74.0

[CMake configuration output](https://github.com/rdiankov/openrave/files/9300984/or-bug.txt).

As a temporary workaround, I renamed VERSION to VERSION.txt.

PeterBowman, 2022-08-10:
This issue has also hit other projects: https://github.com/boostorg/config/issues/345. A common suggestion is to rename or place the version file elsewhere.

cielavenir, 2022-08-11:
GCC11's default is gnu++17 right? why 20 is used?

for now can you try `-DCMAKE_CXX_STANDARD=17`?