rdiankov/openrave #1157: Is there a way to make a docker file for Ubuntu 18.04 bionic?

Eby23, 2022-10-17:
I have tried following lots of issues related to making a docker file. However I am unable to find a good solution. There was a github page which showed on how to make a docker with a lot of bash scripts, but they generate a lot of issues. I want to use openrave and moveit to produce ikfast.cpp files. Any leads would be grateful.

zhubeilife, 2022-10-18:
https://hub.docker.com/r/hamzamerzic/openrave

maybe you can try google with "openrave docker" key words

Eby23, 2022-10-18:
@zhubeilife  the one given by Hamzamerzic is ubuntu:xenial I believe. When I try to follow the steps in the overview, however a lot of error pops up. I want to know if somebody has made a a docker for it.

zhubeilife, 2022-10-19:
@Eby23 maybe you can try this step https://robots.uc3m.es/installation-guides/install-openrave.html#install-openrave-via-scripts-ubuntu-1804-bionic-and-ubuntu-2004-focal