rdiankov/openrave #1307: how to install and use openrave in python3

WenzheHe, 2023-09-11:
I tried to install openrave in python3.6.9, but always failed. Just a few days ago, I saw someone said that now openrave could be install in python3 in another issue, but I didn't any offcial documentation. Could anyone show me how to install openrave in python3? Thanks for help!!

amin-majdi, 2023-12-04:
I have the same question. @WenzheHe did you find any answer for this? I want to install openrave on ubuntu 22.04. I assume you did try the same thing, right?

cielavenir, 2023-12-05:
Now github action is implemented. It targets Ubunutu 20, but I remember Ubuntu 22 installation is quite similar.

aminmajdi, 2023-12-05:
@cielavenir could you please elaborate more. a link to " github action " and the installation guide would be super helpful. thanks

cielavenir, 2023-12-05:
Please read https://github.com/rdiankov/openrave/blob/master/.github/workflows/ci.yaml

WenzheHe, 2023-12-09:
@cielavenir I just follow the installation guide you provided a few days ago. Unfortunately, I met several problems, which I list below:
[1]python3 -m pip install IPython==8.14.0 # ERROR: Could not find a version that satisfies the requirement IPython==8.14.0, I install IPython==8.13.0 instead.
[2]when installing FCL and running ninja -j4, error occurs:
 /usr/include/octomap/OcTreeDataNode.hxx:88:8: note: declared here
   88 |   bool OcTreeDataNode<T>::childExists(unsigned int i) const {
      |        ^~~~~~~~~~~~~~~~~
../src/broadphase/broadphase_dynamic_AABB_tree.cpp:397:50: error: ‘const OcTreeNode’ {aka ‘const class octomap::OcTreeNode’} has no member named ‘getChild’
  397 |         const OcTree::OcTreeNode* child = root2->getChild(i);
      |                                                  ^~~~~~~~
[3]when installing openrave  and running ninja -j4, error ouuurs:
    ../python/bindings/pyann.cpp:317:32: warning: comparison of integer expressions of different signedness: ‘pybind11::size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
	  317 |     BOOST_ASSERT(len(qarray[0])==kdtree.theDim());
	      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
	[454/596] Linking CXX executable src/openrave0.135
	FAILED: src/openrave0.135 
	: && /usr/bin/c++    -rdynamic src/CMakeFiles/openrave.dir/openrave.cpp.o  -o src/openrave0.135  -Wl,-rpath,/home/hwz20/103-git/openrave/build/src/libopenrave-core:/home/hwz20/103-git/openrave/build/src/libopenrave:  src/libboost_assertion_failed.a  /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0  -lrt  -llog4cxx  src/libopenrave-core/libopenrave0.135-core.so.0.135.0  src/libopenrave/libopenrave0.135.so.0.135.0  /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0  /usr/lib/x86_64-linux-gnu/libboost_atomic.so.1.71.0  -lpthread  /usr/lib/x86_64-linux-gnu/libxml2.so  /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0  -ldl  -lrt  -llog4cxx  -lcurl  -lpcre  -lpcrecpp  -lassimp  -lz && :
	/usr/bin/ld: src/libopenrave-core/libopenrave0.135-core.so.0.135.0: undefined reference to `Assimp::aiAssertViolation(char const*, char const*, int)'
	collect2: error: ld returned 1 exit status
and I tried to install openrave with cmake .. and sudo make install, similar error occurs:
    # make error
    #/usr/bin/ld: libopenrave-core/libopenrave0.135-core.so.0.135.0: undefined reference to `Assimp::aiAssertViolation(char const*, char const*, int)'
	collect2: error: ld returned 1 exit status
	make[2]: *** [src/CMakeFiles/openrave.dir/build.make:94: src/openrave0.135] Error 1
	make[1]: *** [CMakeFiles/Makefile2:1235: src/CMakeFiles/openrave.dir/all] Error 2
	make: *** [Makefile:130: all] Error 2
I test this installation guide in Ubuntu-20.04, Python-3.8.10, GCC-9.4.0.

WenzheHe, 2023-12-25:
@aminmajdi hello, have you successfully installed openrave with python 3 and Ubuntu20?

cielavenir, 2023-12-26:
why don't you use docker image: https://hub.docker.com/repository/docker/cielavenir/openrave/tags?page=1&ordering=last_updated

source: https://github.com/cielavenir/dockerfiles/tree/master/openrave

WenzheHe, 2023-12-26:
> why don't you use docker image: https://hub.docker.com/repository/docker/cielavenir/openrave/tags?page=1&ordering=last_updated
> 
> source: https://github.com/cielavenir/dockerfiles/tree/master/openrave

let me try

adriencholet, 2024-02-16:
@WenzheHe Have you been able to install OpenRave with Ubuntu 22.04? I am also having issues and can't seem to find a solution