rdiankov/openrave #1404: Build failure

BlueNoise, 2024-07-22:
It seems that the 'production' branch is broken for building. 'production-stable' seems to be fine.

I can see in CMake:  
```
4.410 CMake Error at src/libopenrave-core/CMakeLists.txt:90 (install):
4.410   install TARGETS given no LIBRARY DESTINATION for shared library target
4.410   "libopenrave-core".
```

I can also see:
```
4.289 CMake Error at src/libopenrave/CMakeLists.txt:73 (install):
4.289   install TARGETS given no LIBRARY DESTINATION for shared library target
4.289   "libopenrave".
```

The lines that are causing problems seem to be, respectively:
```
install(TARGETS libopenrave-core
  EXPORT openrave-targets
  INCLUDES DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}/${OPENRAVE_INCLUDE_INSTALL_DIR}
) 
```

and 

```
install(TARGETS libopenrave
  EXPORT openrave-targets
  INCLUDES DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}/${OPENRAVE_INCLUDE_INSTALL_DIR}
)
```

It seems the problem is there are no DESTINATION to install the targets, as described.

cielavenir, 2024-07-22:
seems cmake 3.14 is required.

BlueNoise, 2024-07-23:
I'll try installing it and see if it works. I'll post my results here after.

BlueNoise, 2024-07-23:
@cielavenir That seems to have done it. Thank you so much.

closed 2024-07-23 by BlueNoise